couin3/RFLink #4: Plugin-30, AlectoV1 does not work

couin3, 2020-03-27:
No message on serial console.
Issue opened to gather informations on this subject.

zoomx, 2020-03-28:
I have added some prints inside the plugin_030 and checked on a Wemos mini against a RFlinkv48 on a Mega.
Receivers are the same, RXB6
Some packets are simply lost but others start the decoding process with the right number of pulsecount, 74, as written at the beginning of the pluging.
But every time the plugin exit on line 115
if (bitstream == 0) return false;          // Perform a sanity check

couin3, 2020-03-28:
Alectov1 seems to have 66 (code) + 8 (checksum) *raw* pulsecount.
Xiron is protocol 046 (shared with Auriol v2) also uses the same *raw* pulsecount.
`#define AURIOLV2_PULSECOUNT 74`

couin3, 2020-03-28:
It seems we never enter in the long pulse part code
`RawSignal.Pulses[x]*RawSignal.Multiply > 2560`

This can mean two things :

1. Protocol is false there : a long pulse is not that long (=> reduce 2560)
2. Core is false (=> `Pulse[x]` timing or `Multiply` is wrong)

couin3, 2020-03-28:
[rtl_433 for Alecto](https://github.com/merbanan/rtl_433/blob/master/src/devices/alecto.c)

> PPM with pulse width 500 us, long gap 4000 us, short gap 2000 us, sync gap 9000 us.

=> Protocols seems to be right (yet, that's some long duration)

couin3, 2020-03-28:
Rised Timeout value (will help for that long duration gap)
`#define SIGNAL_END_TIMEOUT_US        3800`
Can go higher as it will disturb Plugins 046 (Xiron).

@zoomx : If still not working, please test again with this value set to 5000.

zoomx, 2020-03-28:
You're right, Xiron is on Plugin46 and not 45.

Tested: 
5000 works, I get some packets but Xiron seems broken
4000 works too but get some Xiron packets

I have to test more since Xiron don't trasmit as often as Alecto.

It seems that it discard rain and temperature/humidity packet but it happens the same with original RFlink.

Thank you very much, it helped me.  I will play with parameters and post findings here.

couin3, 2020-03-28:
4000 is not working with my 6 Xiron temperature/Humidity sensors.
3000 is the minimum, as NewKaku is not working below.

If you can, please give me the minimum working value for your Alecto v1 sensor.

We can then open another issue to dig further on some discarded messages.

zoomx, 2020-03-28:
3850 works too!

Long testing show that 3400 is good too, I wander why the first test at 3600 didn't work, maybe too short.

couin3, 2020-03-28:
I will set value to a "safe" 3500.
Yet, idk why rain and windspeed are not working (it should!).

couin3, 2020-03-28:
I'm starting to think Checksum is not well calculated.
https://github.com/merbanan/rtl_433_tests/blob/master/tests/alectov1/auriol_protocol_v20.pdf

closed 2020-03-28 by couin3