couin3/RFLink #7: Plugin-30: AlectoV1 does not receive all message types

couin3, 2020-03-29:
This is #4 continued.

It seems all message types are not going though this plugin code.

This document helps a lot to understand how it works internally.
https://github.com/merbanan/rtl_433_tests/blob/master/tests/alectov1/auriol_protocol_v20.pdf

I feel LSB nibble handling may not completely work with checksum.
But I will need *a lot* of raw messages (and their meaning, if known) to go on.

One detail may help my search : Is "battery low" message working?

zoomx, 2020-03-29:
Wind direction is wrong, half the real value seen on the console and RFlink. It is very strange because the plugin is exactly the same!

Battery Low. Since the batteries are all good I get all battery sensors level as OK.

Rain data: it seems that packet is not captured not a plugin fail. More testing.

Protocol in PDF is good but I have another sketch, not RFlink, that is a Frankenstein between tho sources. The first one worked well capturing packets (for Alecto and 3 others protocols) but failed on decoding.
https://github.com/hassebjork/WeatherReader/blob/master/arduino/WeatherReader/WeatherReader.ino
The second one was written for a RaspberryPI so was bad in capturing but good on decoding.
https://github.com/yu55/auriol_reader/blob/master/reader/auriol-reader.c
If you compare decoding here and in PDF seems to match

Frankenstein Code is here
https://github.com/zoomx/ArduinoWork167/tree/master/Alecto_Auriol_WeatherStation
Two versions, one that print in readable format, the other one, not complete, that prints in RFlink format. 
After a packet is captured reportSerial() copy data in encodedBits array then is processed by     decodePluviometer() and decodeWindData()

couin3, 2020-03-29:
> Rain data: it seems that packet is not captured
Do you capture raw data (with R48 firmware)?
Rain and "Combined" sensor may be different than we thought on multiple level
- Transmission power
- Message preamble
- CRC

> Battery Low. Since the batteries are all good I get all battery sensors level as OK.
Can you fake it with discharged/NiMH batteries?

> Wind direction is wrong
I've read in pdf document that Weather station is making some kind of average value.
If some messages are discarded/misinterpreted, that could be tricky to untangle.

zoomx, 2020-03-29:
Please tell me how to capture raw data with R48, I didn't find the command.

In these tests the receivers are all RBX6 close togheter to avoid different signal power. Then I check in parallel with many Realterm windows.

Low signal is working, changed battery. But the plugin print battery value only decoding TH packet, RFlinkR48 instead print this value with all packets

About wind direction: since I am checking in parallel and yesterday there was no wind so the direction was the same all the time. My Frankenstein sketch decode it right, I see the same value in the console.
My mistake was the plugin comparison since I am using R48 that is without sources. I am addind an Arduino Nano with RFlink R29 (the last known sources) reduced, only few plugin. So also RF29 print wind direction wrong. Also RF28 doesn't print BAT on Wind packets, only in TH.

couin3, 2020-03-29:
> Please tell me how to capture raw data with R48, I didn't find the command.
>
RFlink R48 is distributed with "RFLink Loader" Software.
[image]

- Open this software
- Choose the right serial port (COMxxx)
- Push "Serial Port Logging button", wait, you should now be "ON"
- Push "Log Mode : Debug", wait, RFDebug should now be "ON"
- Done

couin3, 2020-03-29:
> About wind direction :
>
```
         winddirection = ((bitstream >> 15) & 0x1ff) / 45; // ???
         winddirection = winddirection & 0x0f;
```
It may be some problems of datatypes / negative value handling.
It should give a value from 0 to 7 ...

Best is to focus on getting some real raw data sets linked with their signification.
This way, I will be able to replay and play more freely ;)

zoomx, 2020-03-29:
I am collecting raw pulses and their meaning with timestamp using RFlink48 and my Frankenstein program at the same time.
I have also an sdr radio and can use rtl433 but it get a lot of signals nearby, some from sensor that I don't have, maybe near me or noisy signals.
I will post raw data here with their meaning after a collection and cleaning from others sensor.

About direction. Since it seems that R29 print direction *2 I believe that there is a shift.

couin3, 2020-03-29:
> I have also an sdr radio and can use rtl433 but it get a lot of signals nearby
>
You can for rtl433 to listen for some protocols only. Check `-R 16`, it should be only AlectoV1.

zoomx, 2020-03-29:
Here is a zip with packets

Auriol_Alecto_rawData.txt are only Auriol-Alecto packets

20200329_RFlink48.txt is the original file

20200329_rflinkEsp.txt is the one captured by your rflink port on ESP8266

20200329_Auriol.txt is the one captured by my frankenstine sketch

With debug on RFlink48 doesn't capture many WINDIR packets, you can see the difference in the original file
Xiron packets seems to be 511 Pulses. Very strange.

As you can see right packet is followed by shorter packet maybe due to serial.print
[RawPAckets.zip](https://github.com/couin3/RFLink/files/4399357/RawPAckets.zip)

zoomx, 2020-03-29:
I made a test with rtl_433
-R 16 works
-A enables the raw signal output but I was not able to capture it
So I used this
rtl_433 -R16 -A >AuriolAlectoRTL433Capture.txt 2>&1
and got this
[AuriolAlectoRTL433Capture.zip](https://github.com/couin3/RFLink/files/4399573/AuriolAlectoRTL433Capture.zip)

couin3, 2020-03-29:
I asked for data, then I got some ;)
I have roughly sorted/cleaned/merged all files.
Can I suppose "Frankenstein", 20200329_Auriol.txt, file contains good translated data? 
What is that second value after Windspeed?
```
"29/03/2020 17:09:18";Auriol;W; 1.00; **0.20**
"29/03/2020 17:09:18";Auriol;WD; 0; 0.60
```

> Xiron packets seems to be 511 Pulses. Very strange.
>
It's somehow normal Xiron behaviour.
The 10(?) messages burst is so fast that it get seperated afterwards

zoomx, 2020-03-29:
> Can I suppose "Frankenstein", 20200329_Auriol.txt, file contains good translated data?

I have to check better the wind values since you're right, the console show average and gust wind.
I have to add that rain is the amount of rain since the first start of pluviometer, when I inserted the batteries. The console made calculation to show millimeters in a day.

> What is that second value (0,20) after Windspeed?
In original sketch I found that values was divided by 5 so I print the first value and then the value divided by 5 to mach easily console values but since console show average I have to check with the PDF.

You made a great work, RFLink Raw Message Injector is a very good idea!

Merci!

couin3, 2020-03-29:
> You made a great work, RFLink Raw Message Injector is a very good idea!
>
Except the name perhaps ;)
It will be of great use with your raw messages and doing some less theoretically debugs.

zoomx, 2020-03-30:
Checked the PDF.
Wind value is called AverageSpeed and it's a 8bit value that should be multiplicated by 0.2 to obtain the value in m/s. Dividing by 5 is equivalent to multiply by 0.2

It's the same for wind gust

couin3, 2020-03-30:
Good news, bad news.
I found why rain message is not received.
Rain sensors has a slightly longer long pulse.
You can rise the value in
`#define SIGNAL_END_TIMEOUT_US 4200`
However, rising that same value will stop Xiron message... (remember their special behaviour in raw message, "`511 Pulses`", that was a side effect).

R48 is dealing with that stuff diffently, I have to discover how. 
Meanwhile, tell me if you effectively get Rain message.

couin3, 2020-03-31:
Change Rain value to be R48 compliant.
It now gives 10th of mm instead of 0.25mm steps.

zoomx, 2020-03-31:
Testin now. I catch rain packets!!!!
R48 doesn't, maybe sometimes it capture and decode rain packets but most are lost.

I noticed that rain and TH packets message are not repeated instead wind packets are repeated.

Maybe the reason is that wind packets are divided in two packets, first one with only wind average, second one with direction and gust. The repeat flag is flagged for the fist one, wind average, but then it receive the direction and gust packet so the flag is resetted again.
It happens with R48 too.

couin3, 2020-03-31:
>Testin now. I catch rain packets!!!!
>
Yeah :) 
Please confirm Xiron is not working anymore.

> Wind packets are repeated
>
TH packets are sent in burst => "SignalCRC" code prevent repetition
Rain packets are sent in burst => "SignalCRC" code prevent repetition as well
Winds packet are mixed (Speed + Dir), then sent in burst => "SignalCRC" code is deceived

> WindSpeed(Gust) value
>
To be R48 compliant, both Speed are converted to tenth of kph , 7.2 * Raw Value (7.2 = 0.2 * 3600/1000 * 10) 

> Channel
>
Is there a way to switch channel on sensors?

couin3, 2020-03-31:
Pushed 1cfaecf modification to handle "mixed bursts".

We should make a new evaluation to see what is still not working.
- Negative temperature value
- Windgust/direction
- ...

zoomx, 2020-03-31:
> Pushed [1cfaecf](https://github.com/couin3/RFLink/commit/1cfaecfccb7673bf45baa7acb2d609aad0288210) modification to handle "mixed bursts".

Seems to work good, no packet repeating.

> We should make a new evaluation to see what is still not working.
> 
>     * Negative temperature value
> 
>     * Windgust/direction

Direction is simply, Windgust... I don't know if moving fast the sensor has an influence, maybe I can try with a hairdryer.
Negative temperature is difficult because the sensor is inside the wind sensor, I have no space in my freezer!

zoomx, 2020-03-31:
Direction in Alecto packet is in degrees.
here
[http://www.rflink.nl/blog2/protref](url)
it is written that 
WINDIR=123 => Wind direction (integer value from 0-15) reflecting 0-360 degrees in 22.5 degree steps
but in plugin is written
            winddirection = ((bitstream >> 15) & 0x1ff) / 45;
            winddirection = winddirection & 0x0f;
When Windir is 270 the value should be 270/22.5=12
My frankenstein
"31/03/2020 17:59:16",WD; 270; 0.80
R48
"31/03/2020 17:59:16",20;1E;Alecto V1;ID=00a0;WINDIR=0012;WINGS=001c;WINSP=0015;BAT=OK;
RFlinkEsp
"31/03/2020 17:59:16",20;0A;Alecto V1;ID=00a0;WINDIR=0006;WINGS=001c;BAT=OK;

Now direction is 90/22.5=4
My frankenstein
"31/03/2020 18:00:49",WD; 90; 1.00
R48
"31/03/2020 18:00:49",20;2C;Alecto V1;ID=00a0;WINDIR=0004;WINGS=0024;WINSP=0015;BAT=OK;
RFlinkEsp
"31/03/2020 18:00:50",20;13;Alecto V1;ID=00a0;WINDIR=0002;WINGS=0024;BAT=OK;

Maybe 45 should be 22.5 but windir should be a float and a integer again before printing.
Don't know why R48 duplicates WINSP in windgust and direction packet since the information is in the previous packet.
Windgust is in the same unit as winspeed so the code should be identical.

couin3, 2020-03-31:
#11 is closed. You should have Xiron and AlectoV1 messages together!

> Negative temperature is difficult because the sensor is inside the wind sensor, I have no space in
my freezer!
>
I'm quite confident for negative temperature (I may be wrong). You can skip that one!

>Direction is simply, Windgust... I don't know if moving fast the sensor has an influence, maybe I can try with a hairdryer.
>
I had no real raw message for WindDir/WindGust. I had to forge them. Need to focus on that!

Except MQTT stuff, Do I miss anything else?

zoomx, 2020-04-01:
In r48 when debug id ON it misses many packets, I just checket that it misses a lot of Windir packets.

So I have only two raw packets

"29/03/2020 17:13:26",20;52;DEBUG;Pulses=74;Pulses(uSec)=480,1860,420,1860,420,1890,420,1860,420,1890,390,3870,420,1890,420,3870,420,1890,420,3870,420,3870,420,1890,420,3870,420,3870,420,3870,420,3870,420,1890,420,3870,420,3870,420,1890,420,3870,420,1890,420,1890,390,1890,420,1890,390,3870,390,3870,420,1890,390,1890,420,1890,390,1890,390,1890,420,3870,390,3870,420,1890,390,1890,390,6990;
"29/03/2020 17:13:26",20;53;Alecto V1;ID=00a0;WINDIR=0002;WINGS=002b;WINSP=000e;BAT=OK;

"01/04/2020 08:50:31",20;BB;DEBUG;Pulses=74;Pulses(uSec)=510,1860,420,1860,450,1860,420,1860,450,1860,420,3840,420,1860,450,3840,420,1860,420,3840,450,3840,420,1860,420,3840,420,3840,420,3840,420,1860,420,3840,420,3840,450,3840,420,1860,450,1860,420,1860,450,1860,420,3840,420,1860,420,1860,420,1860,420,1860,420,1860,420,1860,420,1860,420,1860,420,3840,420,1860,420,1860,420,3840,420,6990;
"01/04/2020 08:50:31",20;BC;Alecto V1;ID=00a0;WINDIR=0012;WINGS=0000;WINSP=0000;BAT=OK;

Xiron for me is still missing.

couin3, 2020-04-01:
> Xiron for me is still missing.
>
Just to be sure :
- Did you get all latest sources ?
- Did you make a "project clean" before build ?

couin3, 2020-04-01:
Let's manually convert those two messages!

**1sr message :**
`"29/03/2020 17:13:26",20;52;DEBUG;Pulses=74;Pulses(uSec)=[...]
`
--> 8 values formatted
```
480,1860,420,1860,420,1890,420,1860,
420,1890,390,3870,420,1890,420,3870,
420,1890,420,3870,420,3870,420,1890,
420,3870,420,3870,420,3870,420,3870,
420,1890,420,3870,420,3870,420,1890,
420,3870,420,1890,420,1890,390,1890,
420,1890,390,3870,390,3870,420,1890,
390,1890,420,1890,390,1890,390,1890,
420,3870,390,3870,420,1890,390,1890,
390,6990;
``` --> last pair is sync, can be removed

--> removing all pulses (even values ~400)
--> keeping gaps
```
1860,1860,1890,1860,
1890,3870,1890,3870,
1890,3870,3870,1890,
3870,3870,3870,3870,
1890,3870,3870,1890,
3870,1890,1890,1890,
1890,3870,3870,1890,
1890,1890,1890,1890,
3870,3870,1890,1890,
```

--> Long gaps (~3800) are "1", Short gaps (~1800) are "0"
```
1860,1860,1890,1860, --> 0000
1890,3870,1890,3870, --> 0101
1890,3870,3870,1890, --> 0110
3870,3870,3870,3870, --> 1111
1890,3870,3870,1890, --> 0110
3870,1890,1890,1890, --> 1000
1890,3870,3870,1890, --> 0111
1890,1890,1890,1890, --> 0000
3870,3870,1890,1890, --> 1100
```

--> Remember LSB First ordering for numbers
```
00000101 0110 111 101101000 01110000 1100
-------- ---- --- --------- -------- ----
ID       TYPE STY DIR       GUST     CRC
A0       WIND DIR 045       14       .
```
R48 :
`"29/03/2020 17:13:26",20;53;Alecto V1;ID=00a0;WINDIR=0002;WINGS=002b;WINSP=000e;BAT=OK;`

**2nd message**
`"01/04/2020 08:50:31",20;BB;DEBUG;Pulses=74;Pulses(uSec)=[...]`

--> 8 values formatted
```
510,1860,420,1860,450,1860,420,1860,
450,1860,420,3840,420,1860,450,3840,
420,1860,420,3840,450,3840,420,1860,
420,3840,420,3840,420,3840,420,1860,
420,3840,420,3840,450,3840,420,1860,
450,1860,420,1860,450,1860,420,3840,
420,1860,420,1860,420,1860,420,1860,
420,1860,420,1860,420,1860,420,1860,
420,3840,420,1860,420,1860,420,3840,
420,6990;
``` --> last pair is sync, can be removed

--> removing all pulses (even values ~400)
--> keeping gaps
```
1860,1860,1860,1860,
1860,3840,1860,3840,
1860,3840,3840,1860,
3840,3840,3840,1860,
3840,3840,3840,1860,
1860,1860,1860,3840,
1860,1860,1860,1860,
1860,1860,1860,1860,
3840,1860,1860,3840,
```

--> Long gaps (~3800) are "1", Short gaps (~1800) are "0"
```
1860,1860,1860,1860, --> 0000
1860,3840,1860,3840, --> 0101
1860,3840,3840,1860, --> 0110
3840,3840,3840,1860, --> 1110
3840,3840,3840,1860, --> 1110
1860,1860,1860,3840, --> 0001
1860,1860,1860,1860, --> 0000
1860,1860,1860,1860, --> 0000
3840,1860,1860,3840, --> 1001
```

--> Remember LSB First ordering for numbers
```
00000101 0110 111 011100001 00000000 1001
-------- ---- --- --------- -------- ----
ID       TYPE STY DIR       GUST     CRC
A0       WIND DIR 270       0        .
```
R48 :
`"01/04/2020 08:50:31",20;BC;Alecto V1;ID=00a0;WINDIR=0012;WINGS=0000;WINSP=0000;BAT=OK;`

couin3, 2020-04-01:
To sum up

**1st message :**
```
00000101 0110 111 101101000 01110000 1100
-------- ---- --- --------- -------- ----
ID       TYPE STY DIR       GUST     CRC
A0       WIND DIR 045       14       .
```
R48 :
`"29/03/2020 17:13:26",20;53;Alecto V1;ID=00a0;WINDIR=0002;WINGS=002b;WINSP=000e;BAT=OK;`

**2nd message**
```
00000101 0110 111 011100001 00000000 1001
-------- ---- --- --------- -------- ----
ID       TYPE STY DIR       GUST     CRC
A0       WIND DIR 270       0        .
```
R48 :
`"01/04/2020 08:50:31",20;BC;Alecto V1;ID=00a0;WINDIR=0012;WINGS=0000;WINSP=0000;BAT=OK;`

couin3, 2020-04-01:
Pushed code 3a4aaf0
Wind Dir raw value divided by 22.5
Value then displayed in Hex Format

zoomx, 2020-04-01:
Got this last version, put in new folder and.... now only Xiron and many Alecto are lost! Since it was very strange, tested on two different wemos mini. Same results. More testing anyway maybe I made silly error.

About direction. I don't know why the author throw away the degrees values. Instead it uses only the 16 direction N, NNE, NE, NEE, E.....
Then the value is still an integer.
WINDIR=123 => Wind direction (integer value from 0-15) reflecting 0-360 degrees in 22.5 degree steps

zoomx, 2020-04-01:
An example

"01/04/2020 15:11:30",20;00;Nodo RadioFrequencyLink - RFLink Gateway V1.1 - R48;
"01/04/2020 15:11:48",20;01;Alecto V1;ID=00a0;WINGS=0000;WINSP=000e;BAT=OK;
"01/04/2020 15:11:49",20;02;Alecto V1;ID=00a0;WINDIR=0000;WINGS=001c;WINSP=000e;BAT=OK;
"01/04/2020 15:11:49",20;03;Alecto V1;ID=00a0;WINGS=001c;WINSP=000e;BAT=OK;
"01/04/2020 15:11:49",20;04;Alecto V1;ID=00a0;WINDIR=0000;WINGS=001c;WINSP=000e;BAT=OK;
"01/04/2020 15:11:49",20;05;Alecto V1;ID=00a0;WINGS=001c;WINSP=000e;BAT=OK;
"01/04/2020 15:11:49",20;06;Alecto V1;ID=00a0;WINDIR=0000;WINGS=001c;WINSP=000e;BAT=OK;
"01/04/2020 15:12:19",20;07;Alecto V1;ID=00a0;WINGS=001c;WINSP=000e;BAT=OK;
"01/04/2020 15:12:20",20;08;Alecto V1;ID=00a0;WINDIR=0000;WINGS=000e;WINSP=000e;BAT=OK;
"01/04/2020 15:12:20",20;09;Alecto V1;ID=00a0;WINGS=000e;WINSP=000e;BAT=OK;
"01/04/2020 15:12:20",20;0A;Alecto V1;ID=00a0;WINDIR=0000;WINGS=000e;WINSP=000e;BAT=OK;
"01/04/2020 15:12:20",20;0B;Alecto V1;ID=00a0;WINGS=000e;WINSP=000e;BAT=OK;
"01/04/2020 15:12:20",20;0C;Alecto V1;ID=00a0;WINDIR=0000;WINGS=000e;WINSP=000e;BAT=OK;
"01/04/2020 15:12:24",20;0D;Xiron;ID=0C03;TEMP=00cb;HUM=62;BAT=OK;
"01/04/2020 15:12:51",20;0E;Alecto V1;ID=00a0;WINDIR=0000;WINGS=0024;WINSP=000e;BAT=OK;
"01/04/2020 15:12:51",20;0F;Alecto V1;ID=00a0;WINGS=0024;WINSP=0007;BAT=OK;
"01/04/2020 15:12:51",20;10;Alecto V1;ID=00a0;WINDIR=0000;WINGS=0024;WINSP=0007;BAT=OK;
"01/04/2020 15:12:51",20;11;Alecto V1;ID=00a0;WINGS=0024;WINSP=0007;BAT=OK;
"01/04/2020 15:12:51",20;12;Alecto V1;ID=00a0;WINDIR=0000;WINGS=0024;WINSP=0007;BAT=OK;
"01/04/2020 15:13:21",20;13;Alecto V1;ID=00a0;WINGS=0024;WINSP=0015;BAT=OK;
"01/04/2020 15:13:22",20;14;Alecto V1;ID=00a0;WINDIR=0014;WINGS=0040;WINSP=0015;BAT=OK;
"01/04/2020 15:13:22",20;15;Alecto V1;ID=00a0;WINGS=0040;WINSP=0015;BAT=OK;
"01/04/2020 15:13:43",20;16;Xiron;ID=0C03;TEMP=00cb;HUM=62;BAT=OK;
"01/04/2020 15:13:52",20;17;Alecto V1;ID=00a0;WINGS=0040;WINSP=0015;BAT=OK;
"01/04/2020 15:13:53",20;18;Alecto V1;ID=00a0;WINDIR=0000;WINGS=0015;WINSP=0015;BAT=OK;
"01/04/2020 15:13:53",20;19;Alecto V1;ID=00a0;WINGS=0015;WINSP=0015;BAT=OK;
"01/04/2020 15:13:53",20;1A;Alecto V1;ID=00a0;WINDIR=0000;WINGS=0015;WINSP=0015;BAT=OK;
"01/04/2020 15:13:53",20;1B;Alecto V1;ID=00a0;WINGS=0015;WINSP=0015;BAT=OK;
"01/04/2020 15:13:53",20;1C;Alecto V1;ID=00a0;WINDIR=0000;WINGS=0015;WINSP=0015;BAT=OK;
"01/04/2020 15:14:23",20;1D;Alecto V1;ID=00a0;TEMP=00b2;HUM=56;BAT=OK;
"01/04/2020 15:14:54",20;1E;Alecto V1;ID=00a0;WINGS=0015;WINSP=000e;BAT=OK;
"01/04/2020 15:14:55",20;1F;Alecto V1;ID=00a0;WINDIR=0000;WINGS=0007;WINSP=000e;BAT=OK;
"01/04/2020 15:14:55",20;20;Alecto V1;ID=00a0;WINGS=0007;WINSP=000e;BAT=OK;
"01/04/2020 15:14:55",20;21;Alecto V1;ID=00a0;WINDIR=0000;WINGS=0007;WINSP=000e;BAT=OK;
"01/04/2020 15:14:55",20;22;Alecto V1;ID=00a0;WINGS=0007;WINSP=000e;BAT=OK;
"01/04/2020 15:14:55",20;23;Alecto V1;ID=00a0;WINDIR=0000;WINGS=0007;WINSP=000e;BAT=OK;
"01/04/2020 15:15:02",20;24;Xiron;ID=0C03;TEMP=00cb;HUM=62;BAT=OK;
"01/04/2020 15:15:25",20;25;Alecto V1;ID=00a0;WINGS=0007;WINSP=0007;BAT=OK;
"01/04/2020 15:15:26",20;26;Alecto V1;ID=00a0;WINDIR=0000;WINGS=0007;WINSP=0007;BAT=OK;
"01/04/2020 15:15:26",20;27;Alecto V1;ID=00a0;WINGS=0007;WINSP=0007;BAT=OK;
"01/04/2020 15:15:26",20;28;Alecto V1;ID=00a0;WINDIR=0000;WINGS=0007;WINSP=0007;BAT=OK;
"01/04/2020 15:15:26",20;29;Alecto V1;ID=00a0;WINGS=0007;WINSP=0007;BAT=OK;
"01/04/2020 15:15:26",20;2A;Alecto V1;ID=00a0;WINDIR=0000;WINGS=0007;WINSP=0007;BAT=OK;
"01/04/2020 15:15:56",20;2B;Alecto V1;ID=00a0;WINGS=0007;WINSP=0000;BAT=OK;
"01/04/2020 15:15:57",20;2C;Alecto V1;ID=00a0;WINDIR=0000;WINGS=0000;WINSP=0000;BAT=OK;
"01/04/2020 15:15:57",20;2D;Alecto V1;ID=00a0;WINGS=0000;WINSP=0000;BAT=OK;
"01/04/2020 15:15:57",20;2E;Alecto V1;ID=00a0;WINDIR=0000;WINGS=0000;WINSP=0000;BAT=OK;
"01/04/2020 15:15:57",20;2F;Alecto V1;ID=00a0;WINGS=0000;WINSP=0000;BAT=OK;
"01/04/2020 15:15:57",20;30;Alecto V1;ID=00a0;WINDIR=0000;WINGS=0000;WINSP=0000;BAT=OK;
"01/04/2020 15:16:21",20;31;Xiron;ID=0C03;TEMP=00cb;HUM=62;BAT=OK;
"01/04/2020 15:16:27",20;32;Alecto V1;ID=00a0;WINGS=0000;WINSP=0000;BAT=OK;
"01/04/2020 15:16:28",20;33;Alecto V1;ID=00a0;WINDIR=0000;WINGS=0007;WINSP=0000;BAT=OK;
"01/04/2020 15:16:28",20;34;Alecto V1;ID=00a0;WINGS=0007;WINSP=0000;BAT=OK;
"01/04/2020 15:16:28",20;35;Alecto V1;ID=00a0;WINDIR=0000;WINGS=0007;WINSP=0000;BAT=OK;
"01/04/2020 15:16:28",20;36;Alecto V1;ID=00a0;WINGS=0007;WINSP=0000;BAT=OK;
"01/04/2020 15:16:28",20;37;Alecto V1;ID=00a0;WINDIR=0000;WINGS=0007;WINSP=0000;BAT=OK;
"01/04/2020 15:16:58",20;38;Alecto V1;ID=00a0;WINGS=0007;WINSP=0000;BAT=OK;
"01/04/2020 15:16:59",20;39;Alecto V1;ID=00a0;WINDIR=0000;WINGS=000e;WINSP=0000;BAT=OK;
"01/04/2020 15:16:59",20;3A;Alecto V1;ID=00a0;WINGS=000e;WINSP=0000;BAT=OK;
"01/04/2020 15:16:59",20;3B;Alecto V1;ID=00a0;WINDIR=0000;WINGS=000e;WINSP=0000;BAT=OK;
"01/04/2020 15:16:59",20;3C;Alecto V1;ID=00a0;WINGS=000e;WINSP=0000;BAT=OK;
"01/04/2020 15:16:59",20;3D;Alecto V1;ID=00a0;WINDIR=0000;WINGS=000e;WINSP=0000;BAT=OK;
"01/04/2020 15:17:08",20;3E;Alecto V1;ID=008f;RAIN=0e26;BAT=OK;
"01/04/2020 15:17:29",20;3F;Alecto V1;ID=00a0;TEMP=00b6;HUM=56;BAT=OK;
"01/04/2020 15:17:40",20;40;Xiron;ID=0C03;TEMP=00cb;HUM=62;BAT=OK;
"01/04/2020 15:18:00",20;41;Alecto V1;ID=00a0;WINGS=000e;WINSP=0007;BAT=OK;
"01/04/2020 15:18:01",20;42;Alecto V1;ID=00a0;WINDIR=0000;WINGS=001c;WINSP=0007;BAT=OK;
"01/04/2020 15:18:01",20;43;Alecto V1;ID=00a0;WINGS=001c;WINSP=0007;BAT=OK;
[... 41 lines cut ...]
[... cut ...]
"01/04/2020 15:15:02",20;03;Xiron;ID=0C03;TEMP=00cb;HUM=62;BAT=OK;
"01/04/2020 15:15:57",20;04;Alecto V1;ID=00a0;WINSP=0000;BAT=OK;
"01/04/2020 15:16:21",20;05;Xiron;ID=0C03;TEMP=00cb;HUM=62;BAT=OK;
"01/04/2020 15:17:29",20;06;Alecto V1;ID=00a0;TEMP=00b6;HUM=56;BAT=OK;
"01/04/2020 15:17:40",20;07;Xiron;ID=0C03;TEMP=00cb;HUM=62;BAT=OK;
"01/04/2020 15:18:59",20;08;Xiron;ID=0C03;TEMP=00cb;HUM=62;BAT=OK;
"01/04/2020 15:20:18",20;09;Xiron;ID=0C03;TEMP=00cb;HUM=62;BAT=OK;
"01/04/2020 15:21:37",20;0A;Xiron;ID=0C03;TEMP=00cb;HUM=62;BAT=OK;

couin3, 2020-04-01:
> Got this last version, put in new folder and.... now only Xiron and many Alecto are lost! Since it was very strange, tested on two different wemos mini. Same results. More testing anyway maybe I made silly error.

My mistake, see #15 ! Rollback done on that part.

couin3, 2020-04-01:
> About direction. I don't know why the author throw away the degrees values. Instead it uses only the 16 direction N, NNE, NE, NEE, E.....
> Then the value is still an integer.
> WINDIR=123 => Wind direction (integer value from 0-15) reflecting 0-360 degrees in 22.5 degree steps

It's a related to developer's state of mind.
Perhaps for it's need, he had a simple display that only could show three letters N-NNE-NE-ENE-...
I prefer not degrading value but not wasting space (a long for a byte) either.

https://www.alecto.nl/ws-3500-professioneel-weerstation

`SPECIFICATIONS
Portée mesurage température intérieure:    0°C ~ +50°C (32°F ~ 122°F)     résolution: 0.1°C (0,2°F)
Portée mesurage température extérieure:    -20°C ~ +60°C (-4°F ~ 140°F)    résolution: 0.1°C (0,2°F) Portée mesurage humidité:    20% ~ 90% humidité relative    résolution:    1%
Portée mesurage pression d’air    700hPa ~ 1100hPa  (27,13inHg ~ 31,89inHg)    résolution: 0,1hPa (0,01inHg)
Temps d’alarme:    120    secondes
Alimentation:    6 x 1,5V pile, format AA, alcaline pour l’écran ou      alimentation de réseau: 6V DC - 300mA -     2 x 1,5V pile, format AA, alcaline pour aiguille de vent     2 x 1,5V pile, format AA, alcaline pour pluviomètre
Quantité précipitation:    0 - 9999mm   -    0 - 393,66 inch
Vitesse de vent:    0 ~ 30m/s    0 ~ 108km/h     0 ~ 67mph    0 ~ 58,3knot    0 ~ 11 Beaufort
Fréquence d’émission aiguille de vent:    434MHz    portée jusqu’à  100m (terrain ouvert)
Fréquence d’émission pluviomètre :    434MHz    portée jusqu’à  50m (terrain ouvert)
`

couin3, 2020-04-01:
**To sum up**
- I think that wind direction is given in 22.5° step, but I have no clue
- I will momentary change code to display degree value without change (in Hex)
- We will decide which format to keep later

zoomx, 2020-04-01:
Now it is working.

Direction
Yes, the console only show 16 direction I believe that inside the Wind Direction sensor there are 16 reed or 16 Hall sensors. But others sensors have a potentiometer so they can measure small angles.
This one is made in China and rebranded  as many of cheap weather station, I bought fron Lidl under the Auriol H13726 name.
The RFlink protocol is here
[http://www.rflink.nl/blog2/protref](url)
and it is written that windirection is an integer between 0 and 15, each of one represent 22.5° steps.
Unfortunately RFlink is a format recognized by many controllers (I am testing MyController on a RaspberryPi) so if we change the format I think we will have errors.

My Xiron sensor is a Digoo R8S or R8G (I have both, they have different protocol). One is decoded by RFlinkR48 as Xiron, the other one as Globaltronics.

couin3, 2020-04-01:
Added MQTT messages to plugin 30 1eb361b

> Unfortunately RFlink is a format recognized by many controllers (I am testing MyController on a RaspberryPi) so if we change the format I think we will have errors.
>
That may be a problem... I may change display to stay "R48 compliant".
Or you may tweak MyController ;)

>Digoo
>
I have Digoo R8H, detected as Xiron

couin3, 2020-04-01:
I will stick to http://www.rflink.nl/blog2/protref

- ID=9999 => device ID (often a rolling code and/or device channel number) (Hexadecimal)
- TEMP=9999 => Temperature celcius (hexadecimal), high bit contains negative sign, needs division by 10
- HUM=99       => Humidity (decimal value: 0-100 to indicate relative humidity in %)
- BAT=OK => Battery status indicator (OK/LOW)
- RAIN=1234 => Total rain in mm. (hexadecimal) 0x8d = 141 decimal = 14.1 mm (needs division by 10)
- WINSP=9999 => Wind speed in km. p/h (hexadecimal) needs division by 10
- WINGS=9999 => Wind Gust in km. p/h (hexadecimal)
- WINDIR=123 => Wind direction (integer value from 0-15) reflecting 0-360 degrees in 22.5 degree steps

zoomx, 2020-04-01:
Mistake, not R8G but R8H like yours.

WINGS in packet is in 0.2 m/s units so you have to divide by 5 to have m/s
Then converion require multiplication by 3.6 (m/s to km/h) and then by 10(to be an integer), so 36.

Or a single multiplication by 7.2

couin3, 2020-04-01:
Pushed 4c68398 for "R48 compatible" display.
You may delay that one if you want to test WINDIR values you received!

couin3, 2020-04-01:
> WINGS in packet is in 0.2 m/s units so you have to divide by 5 to have m/s
> Then converion require multiplication by 3.6 (m/s to km/h) and then by 10(to be an integer), so 36.
> 
> Or a single multiplication by 7.2
>
Allready what is computed (???)
` windspeed = (windspeed * 72) / 10;    // to get 10th of kph`

Now windgust is no more in Tenth of kph
`windgust = (windgust * 72) / 100;    // to get kph`

couin3, 2020-04-04:
Made many minor changes on 030.
Please tell me if all is fine!
I will then close this issue

zoomx, 2020-04-04:
Seems good and Xiron too.

closed 2020-04-04 by couin3