couin3/RFLink #16: Add an I2C Screen display support

couin3, 2020-04-10:
Will show latest received radio / error message.
Default antenna pin may change as I2C SCL/SDA are D1/D2 pins on D1 mini board.

zoomx, 2020-04-10:
I believe that on ESP8266 you can change SCL/SDA pins since it is a software I2C.
Wire.begin(int sda, int scl)

couin3, 2020-04-10:
I'd prefer not to change default pin, in order to use D1 mini shield (for instance https://fr.aliexpress.com/item/32801063529.html)

zoomx, 2020-04-10:
I have the same shield so.... better to change antenna pin!

couin3, 2020-04-14:
This is a v1.0 "hidden" feature.
I'm using U8x8 lib, with 8x8 pixels characters.
This is working on my 128x64 display (=> 16x8 char).
64x48 (8x6 char) D1 mini shield will be truncated.

Code is by default desactivated.
To make it work, you had to :
- Uncomment u8g2 line in platformio.ini
- Uncomment #define OLED_Enabled in RFLink.h
- Uncomment #include <U8x8lib.h> in 8_OLED.cpp

OLED SSD1306 has to be connected like this:
- GND <-> GND
- VCC <-> 3v3
- SCK <-> D1
- SDA <-> D2