couin3/RFLink #18: Errors compiling with plugins that have TRANSMIT functionality

popoviciri, 2020-04-16:
Would not compile with any TX plugin enabled. Using Arduino IDE 1.8.12 or Platformio 1.10.0 with Arduino Core for ESP8266 framework. Output:
```
In file included from sketch\5_Plugin.cpp:348:0:
sketch\.//Plugins/Plugin_080.c: In function 'boolean PluginTX_080(byte, char*)':
sketch\.//Plugins/Plugin_080.c:116:20: error: 'InputBuffer_Serial' was not declared in this scope
    if (strncasecmp(InputBuffer_Serial + 3, "FA20RF;", 7) == 0)
                    ^
sketch\.//Plugins/Plugin_080.c:123:49: error: 'str2int' was not declared in this scope
       bitstream = str2int(InputBuffer_Serial + 8);
                                                 ^
sketch\.//Plugins/Plugin_080.c:124:49: error: 'str2cmd' was not declared in this scope
       byte cmd = str2cmd(InputBuffer_Serial + 19); // ON/OFF
                                                 ^
sketch\.//Plugins/Plugin_080.c:125:18: error: 'VALUE_ON' was not declared in this scope
       if (cmd != VALUE_ON)
                  ^
sketch\.//Plugins/Plugin_080.c:148:17: error: 'RawSendRF' was not declared in this scope
       RawSendRF();
                 ^
exit status 1
Error compiling for board Adafruit Feather HUZZAH ESP8266.
```
I started enabling `char InputBuffer_Serial[INPUT_COMMAND_SIZE];` in `RFLink.h` and so on for the subsequent errors, till I got to `RawSendRF` which has this comment `void RawSendRF(void) {                                                    // * DEPRICATED * DO NOT USE *` in `2_Signal.cpp`.

Surely I am missing something obvious. Any pointers?

Thanks very much for this work!
Cheers!

couin3, 2020-04-16:
Hi! 
As stated is release v1.0 info :
> Plugins R29 compatible (receive only)

We are aiming to start re-activate TX part in V3.0.
Help will be appreciated for all incoming tests!

popoviciri, 2020-04-16:
Hmm, just cloned the repo so I missed the release details, thanks for the answer!
I am interested in plugin 80 and waiting for a shipment of smoke detectors. In the meanwhile I am preparing the RF side integration in Home Assistant and I like this one very much.
Been running there idle on a ESP8266 Huzzah (non the feather) and is picking stuff up from the neighbors:
```
20;01;NewKaku;ID=019a0400;SWITCH=2;CMD=ALLOFF;
20;2C;Chuango;ID=ff8aa3;SWITCH=02;CMD=ON;
20;2D;Auriol;ID=A2;TEMP=008a;BAT=OK;
20;2E;Alecto V1;ID=0024;WINDIR=001;WINGS=0004;BAT=OK;
```
So receives from plugins 4 (NewKaku), 62 (Chuango), 45 (Auriol) and 30 (Alecto V1). I can confirm only the output on serial. My MQTT broker is rotating logs so that info is gone. Will try to adjust that and report back more accurately in the dedicated issue 8.
If this is the kind of testing you need, I' more than happy to report back the results.
Thanks again for your work!
Cheers!

closed 2020-04-16 by popoviciri