couin3/RFLink #24: Error in compiling on Atmega328 board

zoomx, 2020-04-29:
In plugin.cpp
this
#include <FS.h>     // To save MQTT parameters

should be this
#if (!defined(AUTOCONNECT_ENABLED) && defined(MQTT_ENABLED))
#include <FS.h>     // To save MQTT parameters
#endif

Or maybe only #if defined(MQTT_ENABLED))

otherwise you get error because this library doesn't exist in this core.

couin3, 2020-04-29:
At first glance, this make sense for v2.0 (master) and v4.0.
I will soon have a more deep look

couin3, 2020-04-29:
In fact, there's no relation with MQTT there ^^.
We are reading protocols.json file only

couin3, 2020-04-29:
This issue should be solved.
"v2.0" is now a stale branch.(and v3.0 soon will)
Please use Master or V4.0 branches for now.

v4.0 beta (Head) tested OK with this settings : 
- Arduino Nano
- RX plugins : 001, 004, 043, 046, 254
- TX plugins : 004
- PLUGIN_MAX 5
- PLUGIN_TX_MAX 1

closed 2020-04-29 by couin3