couin3/RFLink #25: V.2 reset after decoding the first packet on Arduino Nano

zoomx, 2020-04-29:
I have compiled and loaded the V2.0 version on an Arduino Nano. I have only corrected the plugin 047 as described in issue #20 

I disabled most plugins, leaving only these

#define PLUGIN_029 // Alecto V2 (WS3000/DWK2012) 434 MHz.
#define PLUGIN_030 // Alecto V1 (WS3500) 434 MHz.
#define PLUGIN_031 // Alecto V3 (WS1100/WS1200/WSD-19) 433.9 MHz.
#define PLUGIN_032 // Alecto V4

#define PLUGIN_044 // Auriol v3
#define PLUGIN_045 // Auriol
#define PLUGIN_046 // Auriol v2 / Xiron
#define PLUGIN_047 // Auriol v4

I made also the correction described in issue #24 

I get this 

`"29/04/2020 12:40:16",20;00;RFLink_ESP;VER=1.1;
"29/04/2020 12:40:24",20;01;Alecto V1;ID=008C;RAIN=0e42;BAT=OK;

"29/04/2020 12:40:25",20;00;RFLink_ESP;VER=1.1;
"29/04/2020 12:40:29",20;01;Alecto V1;ID=00A0;WINSP=0015;BAT=OK;

"29/04/2020 12:40:29",20;00;RFLink_ESP;VER=1.1;
"29/04/2020 12:40:29",20;01;Alecto V1;ID=00A0;WINDIR=008;WINGS=0000;BAT=OK;

"29/04/2020 12:40:31",20;00;RFLink_ESP;VER=1.1;
"29/04/2020 12:41:00",20;01;Alecto V1;ID=00A0;WINSP=000e;BAT=OK;

"29/04/2020 12:41:00",20;00;RFLink_ESP;VER=1.1;
"29/04/2020 12:41:00",20;01;Alecto V1;ID=00A0;WINDIR=012;WINGS=0002;BAT=OK;

"29/04/2020 12:41:02",20;00;RFLink_ESP;VER=1.1;

"29/04/2020 12:41:07",20;00;RFLink_ESP;VER=1.1;
"29/04/2020 12:41:19",20;01;Xiron;ID=0C03;TEMP=00dd;HUM=62;BAT=OK;
"29/04/2020 12:41:31",20;02;Alecto V1;ID=00A0;TEMP=00e4;HUM=54;BAT=OK;
`

It seems that few milliseconds after decoding the first packet the MCU resets

zoomx, 2020-04-29:
Changed MaxPlugin from 55 to 10, now I have more RAM but get strange results

`                           
                                                   
20;00;RFLink_ESP;VER=1.1;
                                                     
20;01;Alecto V1;ID=00A0;WINSP=0015;BAT=OK;

                                  
                                                                                
                                                                                
20;00;RFLink_ESP;VER=1.1;
                                                     
20;01;Alecto V1;ID=008C;RAIN=0e42;BAT=OK;
                                     

                                                                            
20;00;RFLink_ESP;VER=1.1;
                                                     
20;01;Alecto V1;ID=00A0;WINSP=0015;BAT=OK;
                                    
¨Î®I[_OiàÃ”*ŸÎ(–­(—p(–¯(—óâ^þO€ƒŠãòÓP–
                       
’Š•á÷®I[_OaàÃ”*ƒÎƒ^O(–­(—ø€ƒ®®E\_OaàÃ”*›­Àù/ÿ~ÿ.ù¡ðO/aàÃ”Ë
Àøð‚óB‰S‰d‰u‰Ã”^
                                                       
ÀWÎ®®E\_OaàÃ”*›­Àù/ûÿ.ù¡ðO/aàÃ”ËÀøð‚óB‰S‰d‰u‰Ã”^
        
À3Îƒàóá^þO€ƒ®K[_O`àÃ”Ð%Î&–­&—óèRÿO€ƒ®K[_OaàÃ”ÐÎ&–­&—óíRÿO€ƒ
®E\_OcàÃ”LÎ‹­œ­­­¾­óìRÿO€ƒ‘ƒ¢ƒ³ƒ®E\_OdàÃ”LñÍóáSÿOK­\­m­~­@ƒQƒbƒsƒ
åÍïâóà‚‚ˆî“à à°à„ƒ•ƒ¦ƒ·ƒ'à2à1ƒ ƒ%ì0à5‡$‡$ì0à7‡&‡ ì0à1‹ ‹!ì0à3‹"‹"ì0à5‹$‹&ì0à7‹
&‹ŽŽŽŽììóà‚‚„ƒ•ƒ¦ƒ·ƒ‰á’à‘ƒ€ƒèíóà„ë‘àßœ’!P0@á÷‰í“à“€“€â•à“(€“'•
Û“Ÿ“”é!¿‘¯‘¢Ÿ€
                                                              
‘£Ÿ
                                                                          
²Ÿ
                                                                            
$•ª»QáÀª»¦·ð¦·ˆ™Z•©÷€••¼Í•¡â.ª»ý
                         
Àª»îÿ¢³äõ ð¢³äõfwˆ™”i÷`•p•€••›¬½Ï•¢Ÿ°³ŸÀ£Ÿp
           
$‘²Ÿp
                                                                      
$‘•”Ú!¥Ÿ
                                                                
´Ÿ
                                                                            
¤Ÿ€
                                                                            
‘$•îÿô‘à-	       ”ààø””"ûÜ
                                      
’
                                                                             
20;00;RFLink_ESP;VER=1.1;
                                                     
                            `
Changed USB port, same results.

When I use my sketch to capture I get expected results.

couin3, 2020-04-29:
Hello again!
Can you :
- Disable Alecto V1 plugin
- Disable all but Alecto V1 plugin
And tell me if you reproduce this issue in both cases?

zoomx, 2020-04-29:
I changed the Arduino nano.

    Disable Alecto V1 plugin
    Disable all but Alecto V1 plugin

Unfortunately same result BUT
reset doesn't happen after printing
20;01;Xiron;ID=0C03;TEMP=00df;HUM=62;BAT=OK;

So I disabled also Alecto V4 and it doesn't reset.

couin3, 2020-04-29:
So, problem is narrowed to both those plugins Alecto V1/V4 ?

zoomx, 2020-04-29:
Yes, it seems that.

what is the function of display_Footer()?
It is called after all printing, for example after display_BAT(battery) but it prints nothing

`void display_Footer(void)
{
  sprintf_P(dbuffer, PSTR("%S"), F(";\r\n"));
  strcat(pbuffer, dbuffer);
}`

this add only ";\r\n" at the end in pbuffer?

Because it happens just after printing results. display_Footer() is called at the end also in Xiron plugins and no reset happens.

couin3, 2020-04-29:
You got it right.
pbuffer should not be directly manipulated by plugins directly.
Hence all the display_XXX funcs (there's a standardisation theme too).

Those kind of wild resets may be due to (p)buffer overrun, I've seen that before...

zoomx, 2020-04-29:
Got reset after this
20;44;Auriol;ID=56;TEMP=0001;BAT=OK;

That seems a spurios signal decoded this way

couin3, 2020-04-29:
Running latest v4.0 branch (Head)
I can reproduce the issue with my own TX3 sensor (LaCrosse 046)

couin3, 2020-04-29:
Found out that AVR is more restrictive with string size...
I have some files to consider!

couin3, 2020-04-29:
This issue should be solved.
"v2.0" is now a stale branch.(and v3.0 soon will)
Please use Master or V4.0 branches for now.

v4.0 beta (Head) tested OK with this settings :
- Arduino Nano
- RX plugins : 001, 004, 043, 046, 254
- TX plugins : 004
- PLUGIN_MAX 5
- PLUGIN_TX_MAX 1

closed 2020-04-29 by couin3