couin3/RFLink #26: Flash OK but 404 Not found

z4rn0x, 2020-05-01:
Hello
I'm very much interested in your project. I flashed it on a Wemos D1 mini. I can see the RFLINK_ESP-xxxxx Wifi network. I can connect to it, enter the password (12345678), and try to connect to the webpage (http://172.217.28.1/) but I can't get anything else than an error 404. It's like a SPIFFS is missing, but I can't find anything on your repo. What am I missing?

couin3, 2020-05-01:
Hi !
Can you precise which version you are using?
Could you also make another with platformIO.ini build flag uncommentend?

[image]

z4rn0x, 2020-05-01:
Here is what I have done so far:
- I tried the "v4.0 Beta1 - Working TX & Webserver"
- I compiled it with Arduino IDE 1.8.12. No error, upload went fine too. I can see and connect to the wifi network ( RFLINK_ESP-xxxxx). When I enter the webpage address 172.217.28.1, I get an error 404
- Following your recommendation, I compiled and uploaded the same project with Platform.io 4.3.3 (first time I use it). Same, no problem to compile and upload, but same error 404
- I tried with RFLink-2.0, and same again
- As I was trying to connect to the wifi with my laptop, I tried to connect with my cellphone, and it worked! Then  I noticed that the full web address was 172.217.28.1/_ac/config (my cellphone automatically came to this address, not my laptop)
- I entered this full address on my laptop and I've been able to do the setup

So, in short, the problem was the web address that my laptop had to reach.

Now it's ok, I have setup my wifi network, my mqtt server, and connected my RXB6 to pins 5-6-7-8 per your instructions:

Name | D1 mini | RXB6
PIN_RF_TX_VCC | D5 | 5 VCC
PIN_RF_TX_NA | D6 | 6 DER
PIN_RF_TX_DATA | D7 | 7 DAT
PIN_RF_TX_GND | D8 | 8 GND

IN RF LINK Settings, I setup the GPIOs:
Receiver:
RX_PMOS (empty)
RX_NMOS (empty)
RX_VCC 5
RX_GND 8
RX_NA 6
RX_DATA 7
Pullup on RX_DATA unchecked

Is it ok? I can't see any message yet, I don't have a 433Mhz sensor, but I was hoping my neighbors do...

couin3, 2020-05-01:
Hello again,
Until now, I used my smartphone for this initial setup.
I will keep in mind the behavior is different when using a laptop.
Thanks a lot for all those tries, especialy if you don't have any 433Mhz sensors... yet ;)

z4rn0x, 2020-05-01:
Good and bad news... As I wasn't receiving any data, I tried the "official" RF LINK on a Mega2560 and I received some info:
20;03;Oregon Temp;ID=4E10;TEMP=00d4;BAT=OK;
20;04;Oregon Temp;ID=4E10;TEMP=00d5;BAT=OK;
20;05;Oregon Temp;ID=4E10;TEMP=00d5;BAT=OK;
20;06;AcuriteV2;ID=b795;TEMP=00cf;HUM=78;BAT=OK;
20;07;AcuriteV2;ID=b795;TEMP=00ce;HUM=78;BAT=OK;
20;08;Oregon Temp;ID=4E10;TEMP=00d4;BAT=OK;
20;09;AcuriteV2;ID=b795;TEMP=00ce;HUM=77;BAT=OK;
20;0A;Oregon Temp;ID=4E10;TEMP=00d4;BAT=OK;
20;0B;AcuriteV2;ID=b795;TEMP=00cb;HUM=75;BAT=OK;
20;0C;Oregon Temp;ID=4E10;TEMP=00d3;BAT=OK;

So my hopes have been satisfied... Question now: what is wrong with my RFLINK ESP? Any idea?

couin3, 2020-05-01:
That's a different subject!
Quick answers :
- Your may activate one ESP Debug mode to check whether receiver sees thoses *raw* message
- RFLink ESP is based on "original " RFLink R29, we don't have plugins for "AcuriteV2" nor "Oregon Temp" yet.

closed 2020-05-03 by z4rn0x