couin3/RFLink #27: General comments

chm26, 2020-05-02:
Thank you for the work.
I don't know if it is the right way to use a git issue,  but I would like to make the following remarks : 
- in README.md §7 : are you sure it is PIN_RF_TX_xxx and not  PIN_RF_RX_xxx instead ?
- it will be nice to have, in the serial output, a log that gives the current PIN_RF_RX_DATA value (in fact I had some difficulties to identify this PIN, especially when I disable AUTOCONNECT_ENABLED).
- By default, I had a problem with generated file '.vscode\c_cpp_propertires.json' because in the 'includePath' it can't find the "xxxx/RFLink/include". I don't know if it is a good solution, but in the platformio.ini, I add 'include_dir = RFLink'. Perhaps, another solution is to add an 'include' directory in git.

Sincerely.

couin3, 2020-05-02:
Hi !
- in README.md §7 : are you sure it is PIN_RF_TX_xxx and not PIN_RF_RX_xxx instead ?
Readme has not been updated since v1.0, Pinout section has to be updated there.

- it will be nice to have, in the serial output, a log that gives the current PIN_RF_RX_DATA value (in fact I had some difficulties to identify this PIN, especially when I disable AUTOCONNECT_ENABLED).
If Autoconnect is active, Pin settings is on Webserver (default : no pin active)
If Autoconnect is inactive, Pin settings is fixed in 1_Radio.h
However, It's a nice idea to Serial output active pins.

- By default, I had a problem with generated file '.vscode\c_cpp_properties.json' because in the 'includePath' it can't find the "xxxx/RFLink/include". I don't know if it is a good solution, but in the platformio.ini, I add 'include_dir = RFLink'. Perhaps, another solution is to add an 'include' directory in git.
I encountered that one recently. May be a change in PlatformIO. I will dig further.

couin3, 2020-05-02:
Default wiring (without Autoconnect) is still subject to change!
I close this issue for now.

closed 2020-05-02 by couin3