couin3/RFLink #40: Juste un peu d'aide

MatthBou, 2020-11-22:
Bonjour,

je me suis lancé sur ce projet super intéressant. Je t'ai contacté sur le forum de Domoticz mais je me permet d'écrire ici même si je sais que ce n'est pas forcement le bon endroit pour demander de l'aide...

j'ai deux problemes

1) 
Je reçois bien le message en MQTT au démarage du D1 :
"Online"
"20;00;RFLink_ESP;VER=5.1;↵"

mais je n'arrive pas a recevoir de données en 443, c'est probablement un problème de câblage.
j'utilise un SRX882.
je ne comprend pas cette partie du code :
#define PIN_RF_RX_PMOS_0 NOT_A_PIN // High Side P-MOSFET, active on LOW level
#define PIN_RF_RX_NMOS_0 D5 // Low Side N-MOSFET, active on HIGH level
#define PIN_RF_RX_VCC_0 NOT_A_PIN // Power to the receiver on this pin
#define PIN_RF_RX_GND_0 NOT_A_PIN // Ground to the receiver on this pin
#define PIN_RF_RX_NA_0 NOT_A_PIN // Alt. RX_DATA. Forced as input
#define PIN_RF_RX_DATA_0 D6 // On this input, the 433Mhz-RF signal is received. LOW when no signal.
#define PIN_RF_TX_PMOS_0 NOT_A_PIN // High Side P-MOSFET, active on LOW level
#define PIN_RF_TX_NMOS_0 D7 // Low Side N-MOSFET, active on HIGH level
#define PIN_RF_TX_VCC_0 NOT_A_PIN // +5 volt / Vcc power to the transmitter on this pin
#define PIN_RF_TX_GND_0 NOT_A_PIN // Ground power to the transmitter on this pin
#define PIN_RF_TX_NA_0 NOT_A_PIN // Spare RX pin. Forced as input
#define PIN_RF_TX_DATA_0 D4 // Data to the 433Mhz transmitter on this pin

Qu'est ce que le mosfet que je dois connecter en D5? pourrais tu m’aiguiller sur le câblage?

2)
Par ailleurs je n'arrive pas a me connecter au serveur web, c'est une fonctionnalité qui n'est pas active?

Merci d'avance de ton aide.
Matthieu

couin3, 2020-11-22:
Hello,
Je crois que le 1er problème est simple, il faut refaire le guide d'installation ^^.
De mes petites expériences, j'ai retenu qu'on peut avoir des soucis pour contrôler directement un couple Emetteur / récepteur avec un ESP8266. D'où la solution de passer par l'intermédiaire de MOSFET dans ce cas.
Les Arduino Mega et ESP32 n'on pas ce problème! Il y a donc du paramétrage pour jouer avec un peu toute les configurations possibles!
A vrai dire, les différentes combinaisons hardware sont un vrai casse tête (si on veut tout tester à chaque mise à jour).
C'est pour ça que je recommande en particulier l'ESP32 avec un RXB6, car c'est celui que je maintiens actuellement.

Bref... Il faut tester si ton SRX882 fonctionne déjà avec un RFLink original (sur Arduino Mega).
Alternativement, tu peux modifier les configurations Pin RF_RX_xxx en t'aidant des commentaires, pour respecter le brochage du SRX882.

couin3, 2021-01-02:
N'hésite pas à rejoindre la discussion Discord https://discord.gg/UryRx2hpbm

closed 2021-01-02 by couin3