couin3/RFLink #42: Problem reading data from TCM97001 Sensor

Dattel, 2020-12-29:
Hi,
as mentioned in my recent issue, i would like to add a new sensor definition. The Sensor is called something like "TCM97001".
It also transmits it's data in a 36BIT stream - i managed to decrypt the datafields with another RF-Implementation. Nevertheless, RFLink does not seems to pickup these data-stream...

AAAA  BBBB CCCC DDDD EEEE FFFF GGGG HHHH IIII JJJJ
A=Startbits -> always 1001 ??
B+C= Random Address
D= 1B Battery, 1B Manually trasmittion, 2B Channel
EFG= 12 bit temperature  (last2 bits indicates negative value)
HI = Humidity

I picked up some sample data with these [implementation of RX-Receiver](https://forum.arduino.cc/index.php?topic=136836.45):

```
Start Bit L:   8580   H: 656
Data Bits: 36

L: 3755 1812 1812 3794 1834 1814 1814 1883 1828 1816 3837 1910 1847 1836 1835 3818 1869 1822 1827 1885 3772 3732   3756 1884 1872 3728 1824 3825 1869 1832 1909 1892 1861 1918 1828 1984

H: 558 570 568 567 568 570 566 567 573 566 547 540 556 540 548 543 532 563 550 567 542 557 539 566 532 565 554 560   536 548 563 559 540 552 551 559

```
- 36 bits
- highbits seems to be always around 550us
- lowbits are above 3500us
- the preamble is 8580us

I already tried to activate debugging for Plugin_001 and Plugin_254 to see something similar to my already known definition - but the received streams does not match... 

Can anyone help with that?

By comparing some other implementations for RF-Receiving i stumbled across a main difference... Your implementation queries the RF-Receiver in a loop, while other implementations uses an interrupt - can you explain, why one method is better than the other?

Dattel, 2021-01-09:
is there something i can provide, to help identifiying these sensor?

cpainchaud, 2021-02-12:
Hi! Can you still provide the output  with command 10;rfdebug=on; ?

Dattel, 2021-02-12:
https://manuall.de/auriol-ian-279818-wetterstation/
https://www.ebay.de/itm/Temperaturstation-Wetterstation-Funkuhr-Aussensensor-Temperaturanzeige-Funksensor-/263383205271

As i remember, there is only crap comming in with rfdebug=on using these sensor... the buffer is running full and only posts rubbish..

Here is my sample implementation, from https://forum.arduino.cc/index.php?topic=136836.45, which is able to detect the signal and all of it's used bits and timings. 

```
/*
  Sketch zur Vorab-Analyse unbekannter 433-MHZ-Wettersensoren
  und Fernbedienungen von 433MHz-Funksteckdosen
  Inspiriert durch Beiträge im Arduino-Forum:
  http://arduino.cc/forum/index.php/topic,119739.0.html
  http://arduino.cc/forum/index.php/topic,136836.0.html
 
  Hardware:
  1. Arduino-Board mit 433 MHz Regenerativempfänger für ASK/OOK,
  angeschlossen an einem interruptfähigen Pin.
  2. Funksensor entweder eines 433 MHz Funkthermometers
  oder Funk-Wetterstation oder Steckdosen-Funkfernbedienung
 
  Analysiert werden können Sensoren mit folgendem Funkprotokoll:
  - extra langes Startbit (extra langer LOW Pulse am Receiver)
  - langes 1-Datenbit  (langer LOW Pulse am Receiver)
  - kurzes 0-Datenbit  (kurzer LOW Pulse am Receiver)
  - sehr kurze Trenn-Impulse zwischen den Datenbits (sehr kurze HIGH-Impulse am Receiver)
  - 20 bis 50 Datenbits pro Datenpaket
  Diese Art Funkprotokoll trifft auf die meisten billigen 433 MHZ
  Funkthermometer, Funk-Wetterstationen und Funk-Steckdosen zu.
 
  Ausgabe ueber den seriellen Monitor
  Je erkanntem Datenpaket am Receiver wird ausgegeben:
  - Länge des Startbits (Mikrosekunden LOW) und des nachfolgenden HIGH-Impulses
  - Anzahl der erkannten Datenbits im Datenpaket
  - Länge aller erkannten Datenbits (Mikrosekunden LOW)
  - Länge der zwischen den Datenbits erkannten Pausen (Mikrosekunden HIGH)
  - die als 0/1-Bitstrom decodierten Datenbits des Datenpakets
 
  Nur Vorab-Analyse des Timings im Funkprotokoll!
  In einem weiteren Schritt muss dann die Bedeutung der Bits
  und die Umsetzung in Messwerte erst noch detalliert decodiert werden,
  dieser Sketch erkennt nur das Timing und die Groesse der Datenpakete!
*/

#include <Arduino.h>

// connect data pin of rx433 module to a pin that can handle hardware interrupts
// with an Arduino UNO this is digital I/O pin 2 or 3 only
#define RX433DATAPIN 5

// hardware interrupt connected to the pin
// with Arduino UNO interrupt-0 belongs to pin-2, interrupt-1 to pin-3
//#define RX433INTERRUPT 0
#define RX433INTERRUPT 5

// Set speed of serial in Arduino IDE to the following value
#define SERIALSPEED 115200

// Now make some suggestions about pulse lengths that may be detected
// minimum duration (microseconds) of the start pulse
#define MINSTARTPULSE 3800

// minimum duration (microseconds) of a short bit pulse
#define MINBITPULSE 450

// minimum duration (microseconds) of a HIGH pulse between valid bits
#define MINHITIME 50

// variance between pulses that should have the same duration
#define PULSEVARIANCE 250
[... 127 lines cut ...]
        shortBitDuration-=PULSEVARIANCE;
      longBitDuration=lowtime*2+PULSEVARIANCE;
      validpulsbuf[counter]=lowtime;
      validhibuf[counter]=hitime;
      counter++;
    }
    else if (counting&&(lowtime>shortBitDuration)&&(lowtime<longBitDuration))
    {
      validpulsbuf[counter]=lowtime;
      validhibuf[counter]=hitime;
      counter++;
      if ((counter==MAXPULSECOUNT) || (hitime<MINHITIME))
      {
        showBuffer();
      } 
    }
    else // Low Pulse is too short
    {
      if (counting) showBuffer();
      counting=false;
      counter=0;
    } 
  }
}
```

cpainchaud, 2021-02-12:
Oh so this is a new Auriol device or at least one that we don't have (Plugins  44, 45, 46 ,47)

The fact that preamble is 8500us could be the issue here and may explain why you are getting such garbage as we stop to drop after 5000us for single pulse. Also this code example you provided mentions very short pulses (50us) so we need to change another value there too:
Can you do a capture again with a change to some values:
#define SIGNAL_END_TIMEOUT_US 15000 (instead of 5000 by default)
#define MIN_PULSE_LENGTH_US 20 (instead of 100)

We also have an new branch based on interrupts which provided better results on 'busy' devices but usually unless you did something like customization there your device should not be "busy".

Dattel, 2021-02-12:
Now i'm working with RFLink-V8 implementation, as you suggested due too the interrupt implementation..

First of all, some general small amendment requests for Auriol:

plugin_46.c line 155 changed, since Auriol also has 3 channels:
`if ((type == 0) && (channel > 3))`

plugin_46.c line 155 line 193 added output for current channel:
`display_CHAN(channel);`

4_Display.h line 54 added stub method for display_CHAN:
`void display_CHAN(byte);`

4_Display.cpp line 198: added method for display_CHAN:
```
// Channel
void display_CHAN(byte channel)
{
  sprintf_P(dbuffer, PSTR("%s%04x"), PSTR(";CHN="), channel);
  strcat(pbuffer, dbuffer);
}
```

Please forget my last post - i mixed something up in my head and guided you to the wrong sensors, which already works with rflink and my suggested changes from above.

The correct sensor to implement is this one:

https://www.ebay.de/p/591828645
https://www.amazon.de/Steinbach-Digitales-Funk-Thermometer-061333/dp/B01N28KKBH
https://encrypted-tbn1.gstatic.com/shopping?q=tbn:ANd9GcRZlUaHQRDOSyYnVwboTeMWGMt3SFvfVeS0_8xdj9GwAWVq4DVewg&usqp=CAc

here are some console outputs with DEBUG=TRUE:
There is a lot of rubbish coming in, but when i insert the battery, some long data is comming in... in know, that these sensor repeats it's temperature multiple times in a short sequence..

```
20;XX;DEBUG;Pulses=291;Pulses(uSec)=205,12154,295,3311,1324,47,2515,3170,4295,1988,1195,1379,1403,122,1851,6367,125,2236,301,3275,2258,594,256,1754,66,2138,141,4447,267,1088,59,3629,516,540,762,1806,712,3647,682,401,2016,1715,1723,841,34,1798,645,1785,665,1743,627,1779,640,1856,610,3697,654,1754,663,1746,628,1844,621,3711,606,1799,607,1800,635,3764,615,1817,594,1837,612,1791,601,1873,621,3686,588,3746,597,3732,606,1862,636,1822,605,1797,606,3730,589,1884,572,1813,609,1821,582,1824,588,1904,599,1832,588,1820,589,1816,586,1909,563,918,669,8509,593,3761,574,1831,573,1835,576,3819,596,1834,584,1826,587,1842,571,1905,550,3758,584,1824,578,1832,587,1886,578,3778,571,1832,570,1840,572,3828,569,1859,578,1821,577,1860,570,1904,545,3764,579,3757,565,3790,571,1902,568,1837,569,1841,579,3754,572,1898,573,1861,565,1839,569,1836,579,1877,587,1824,580,1821,585,1829,578,1893,616,863,720,8530,614,3742,542,1860,572,1839,570,3828,587,1843,580,1850,566,1864,575,1921,586,3768,566,1842,567,1829,563,1909,542,3765,574,1836,563,1846,564,3832,567,1862,570,1860,561,1848,569,1903,551,3760,569,3760,579,3746,568,1927,576,1831,568,1838,576,3755,575,1898,634,1799,569,1838,568,1862,563,1910,542,1843,566,1841,571,1834,565,1931,577,949,670,8547,570,3784,572,1837,571,1836,574,3826,574,1832,567,1843,567,1839,563,1912,569,3785,569,1836,570,1836,582,1918,589;
20;0F;DEBUG;
20;XX;DEBUG;Pulses=202;Pulses(uSec)=564,1841,559,1847,564,3835,568,1858,603,1805,561,1866,568,1908,551,3756,565,3772,579,3749,582,1915,572,1835,574,1836,573,3758,587,1881,598,1835,573,1857,560,1850,555,1918,536,1842,595,1816,563,1861,591,1889,550,929,640,8530,579,3759,555,1873,556,1873,590,3789,560,1842,589,1815,580,1822,591,1890,562,3773,559,1861,572,1863,562,1912,534,3776,563,1843,585,1823,560,3834,591,1838,582,1847,602,1810,559,1910,562,3750,578,3756,557,3776,560,1924,553,1866,575,1832,555,3774,565,1911,562,1887,558,1851,570,1830,583,1921,562,1866,565,1842,557,1850,561,1913,547,933,649,8551,564,3788,560,1848,567,1839,565,3835,569,1862,569,1838,575,1853,566,1906,563,3748,563,1838,588,1818,595,1884,561,3771,570,1881,569,1832,587,3819,587,1841,566,1863,563,1886,572,1904,564,3748,577,3755,563,3769,567,1931,570,1829,573,1842,569,3760,564,1910,582,1851,566,1861,563,1843,573,1902,544,1842,559,1868,565,1843,559,1915,549,471,548,1850;
20;XX;DEBUG;Pulses=291;Pulses(uSec)=589,685,196,2359,241,191,202,805,215,3387,67,5961,32,234,34,107,203,7148,256,1256,332,705,82,86,2270,3103,143,5768,457,1516,718,2072,44,359,54,25,1163,2740,135,3328,153,2210,174,550,44,176,265,6073,23,76,803,1175,235,5995,1502,2337,1829,214,34,1265,689,1705,676,3724,658,1762,652,1741,640,1800,626,1882,651,1764,614,189,47,3435,629,3697,625,1838,585,3690,631,1816,619,1801,618,3753,602,1815,600,1838,608,1829,612,1896,610,3697,598,3705,576,3707,601,1862,596,3711,585,1808,591,1800,591,3783,595,1798,600,1795,587,1806,584,1877,564,1808,566,1823,597,1797,617,1845,582,931,691,8449,663,3651,572,1810,560,1836,594,3799,587,1831,566,1834,584,1831,581,1880,580,1842,558,3724,575,3707,582,1880,580,3750,576,1838,579,1815,579,3774,558,1813,557,1838,595,1844,581,1904,551,3733,591,3740,574,3748,631,1836,572,3732,592,1804,561,1836,574,3819,672,1746,562,1834,655,1783,604,1884,547,1824,554,1840,562,1867,569,1940,570,938,724,8425,568,3738,575,1814,564,1836,573,3821,572,1846,559,1836,578,1837,576,1889,572,1844,547,3734,572,3712,583,1877,588,3743,570,1846,572,1821,580,3771,556,1817,574,1818,588,1854,572,1935,570,3738,553,3730,588,3689,633,1833,574,3757,567,1849,570,1817,582,3774,555,1816,550,1847,572,1867,588,1919,563,1832,574,1820,604,1792,570,1890,579,955,660,8465,565,3742,566,1831,564;
20;6B;DEBUG;
20;XX;DEBUG;Pulses=226;Pulses(uSec)=555,1830,567,1830,563,3787,578,1786,581,1825,560,1880,557,1943,582,1817,559,3748,576,3750,569,1893,560,3735,603,1795,582,1818,574,3825,560,1857,545,1845,584,1861,574,1931,578,3730,535,3748,568,3717,570,1891,566,3762,576,1839,603,1795,570,3777,577,1797,574,1814,591,1808,549,1917,569,1824,566,1828,563,1829,571,1892,571,961,652,8471,572,3757,567,1852,577,1816,569,3778,562,1816,565,1827,573,1820,557,1909,591,1826,540,3749,569,3751,581,1885,573,3729,574,1820,542,1856,592,3802,563,1855,560,1834,569,187 [... cut ...] 35,552,3736,605,3701,571,1913,547,3734,580,1858,577,1843,570,3803,572,1844,576,1865,570,1845,573,1888,570,1845,579,1818,569,1870,567,1931,577,944,680,8466,590,3710,576,1825,632,1757,593,3762,555,1814,562,1833,574,1869,569,1937,570,1847,545,3739,566,3759,570,1892,590,3714,578,1817,578,1815,568,3830,576,1841,573,1822,575,1817,543,1918,578,3729,541,3741,569,3712,594,1860,605,3730,588,1852,569,1848,572,3800,576,1841,574,1868,580,1836,567,1896,571,1845,570,1824,590,1846,640,1870,557,463,617,1889;
```

cpainchaud, 2021-02-12:
ok i will have a look.

note that V8 will probably not compile very well for you it's a work in progress

Dattel, 2021-02-12:
but v8 has no problems to compile and upload for me....... strange... I'm using ESP32-WROOM-32D

I'm not sure, i there is any benefit for you, if i post my recent analytics on these TCM-Sensor.. I captured some data with the earlier mentions sample code and analyzed it using an excel sheet.

[image]

cpainchaud, 2021-02-12:
did you  compile V8 with flag "RFLINK_ASYNC_RECEIVER_ENABLED" ? this how you turn on "interrupt based" reading

cpainchaud, 2021-02-12:
based on this post : https://forum.fhem.de/index.php?topic=111531.0 your sensor is a Steinbach_Pool CUL_TCM97001 model Prologue

Dattel, 2021-02-12:
:-( 
You are right... didn't realize the flag RFLINK_ASYNC_RECEIVER_ENABLED.... 
thought, that it is enabled by default... No success on build - but not so bad.

yes, that's right... TCM97001....  should be supported since R34 (written in readmeRfLink.txt)... but dosn't work with mine...!?

cpainchaud, 2021-02-12:
yes ... short story this project is a fork from RFlink which is not opensource anymore. That said TCM97001 is used by many vendors and they dont use it the same way so we need to look at it. First we need a clean signal! what you provided earlier is really a mess. what is the other project you used to get a clean signal? and was it with same arduino/ESP hardware?

If you want to try ASYNC_RECEIVER please use my repo (until we commit it here) https://github.com/cpainchaud/RFLink branch=master

Dattel, 2021-02-12:
yeah... i already realized it by reading the description :-D

maybe i have time tomorrow an try the async implementation..

here are again some raw informations (also included in the first post) i captured with these sensor last year to readout data... I used the sample implementation above.

```
Start Bit L:   8580   H: 656

Data Bits: 36
L: 3755 1812 1812 3794 1834 1814 1814   1883 1828 1816 3837 1910 1847 1836 1835 3818 1869 1822 1827 1885 3772 3732   3756 1884 1872 3728 1824 3825 1869 1832 1909 1892 1861 1918 1828 1984

H: 558 570 568 567 568 570 566 567 573   566 547 540 556 540 548 543 532 563 550 567 542 557 539 566 532 565 554 560   536 548 563 559 540 552 551 559
```

you can read it like a concatenation of:
   StartBitL + StartBitH + L1 +H1 +L2 +H2 +L3 +H3....

in my case for these signal:
8580 Low       -> Start sequence
656 High       -> sync

3755 LOW      -> 1
558 HIGH       -> sync
1812 LOW      -> 0
570 HIGH       -> sync
1812 LOW      -> 0
568 HIGH       -> sync
3794 LOW      -> 1
.... and so on

- so the startbit is ~ 8600us
- an 1 is >3500 us
- an 0 is >1500us and <2000us
- the sync is ~550us...

does that help?

Dattel, 2021-02-14:
> If you want to try ASYNC_RECEIVER please use my repo (until we commit it here) https://github.com/cpainchaud/RFLink branch=master

I've got problems to compile...... something wrong with the included tzapu lib.

```
Processing doitESP32 (platform: espressif32; board: esp32doit-devkit-v1; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 (3.0.0) > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 160MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 3.10004.210126 (1.0.4)
 - tool-esptoolpy 1.30000.201119 (3.0.0)
 - toolchain-xtensa32 2.50200.80 (5.2.0)
Converting RFLink.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain+, Compatibility ~ soft
Library Manager: Installing tzapu/WiFiManager @ ^2.0.4-beta
Warning! Could not find the package with 'tzapu/WiFiManager @ ^2.0.4-beta' requirements for your system 'windows_amd64'
Found 27 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <PubSubClient> 2.8.0
|-- <WiFi> 1.0
Building in release mode
Compiling .pio\build\doitESP32\src\10_Wifi.cpp.o
Compiling .pio\build\doitESP32\src\1_Radio.cpp.o
Compiling .pio\build\doitESP32\src\2_Signal.cpp.o
Compiling .pio\build\doitESP32\src\3_Serial.cpp.o
Compiling .pio\build\doitESP32\src\4_Display.cpp.o
Compiling .pio\build\doitESP32\src\5_Plugin.cpp.o
Compiling .pio\build\doitESP32\src\6_WiFi_MQTT.cpp.o
Compiling .pio\build\doitESP32\src\7_Utils.cpp.o
Compiling .pio\build\doitESP32\src\8_OLED.cpp.o
Compiling .pio\build\doitESP32\src\9_Serial2Net.cpp.o
Compiling .pio\build\doitESP32\src\Plugins\Plugin_001.c.o
Compiling .pio\build\doitESP32\src\Plugins\Plugin_002.c.o
Compiling .pio\build\doitESP32\src\Plugins\Plugin_003.c.o
Compiling .pio\build\doitESP32\src\Plugins\Plugin_004.c.o
Compiling .pio\build\doitESP32\src\Plugins\Plugin_005.c.o
Compiling .pio\build\doitESP32\src\Plugins\Plugin_006.c.o
Compiling .pio\build\doitESP32\src\Plugins\Plugin_007.c.o
Compiling .pio\build\doitESP32\src\Plugins\Plugin_008.c.o
Compiling .pio\build\doitESP32\src\Plugins\Plugin_009.c.o
Compiling .pio\build\doitESP32\src\Plugins\Plugin_010.c.o
Compiling .pio\build\doitESP32\src\Plugins\Plugin_011.c.o
Compiling .pio\build\doitESP32\src\Plugins\Plugin_012.c.o
Compiling .pio\build\doitESP32\src\Plugins\Plugin_013.c.o
Compiling .pio\build\doitESP32\src\Plugins\Plugin_014.c.o
Compiling .pio\build\doitESP32\src\Plugins\Plugin_015.c.o
Compiling .pio\build\doitESP32\src\Plugins\Plugin_029.c.o
Compiling .pio\build\doitESP32\src\Plugins\Plugin_030.c.o
Compiling .pio\build\doitESP32\src\Plugins\Plugin_031.c.o
Compiling .pio\build\doitESP32\src\Plugins\Plugin_032.c.o
Compiling .pio\build\doitESP32\src\Plugins\Plugin_033.c.o
Compiling .pio\build\doitESP32\src\Plugins\Plugin_034.c.o
Compiling .pio\build\doitESP32\src\Plugins\Plugin_035.c.o
Compiling .pio\build\doitESP32\src\Plugins\Plugin_036.c.o
In file included from RFLink\5_Plugin.cpp:183:0:
RFLink\./Plugins/Plugin_037.c: In function 'boolean Plugin_037(byte, const char*)':
RFLink\./Plugins/Plugin_037.c:183:46: warning: format '%x' expects argument of type 'unsigned int', but argument 3 has type 'long unsigned int' [-Wformat=]
Compiling .pio\build\doitESP32\src\Plugins\Plugin_037.c.o
    sprintf(c_ID, "%02x%02x", (rc & 0xFF), rc2);
                                              ^
Compiling .pio\build\doitESP32\src\Plugins\Plugin_040.c.o
Compiling .pio\build\doitESP32\src\Plugins\Plugin_041.c.o
RFLink\./Plugins/Plugin_037.c:183:46: warning: format '%x' expects argument of type 'unsigned int', but argument 3 has type 'long unsigned int' [-Wformat=]
In file included from RFLink\5_Plugin.cpp:339:0:
RFLink\./Plugins/Plugin_076.c: In function 'boolean Plugin_076(byte, const char*)':
[... 36 lines cut ...]
Compiling .pio\build\doitESP32\lib34e\WiFi\WiFiClient.cpp.o
Compiling .pio\build\doitESP32\lib34e\WiFi\WiFiGeneric.cpp.o
Compiling .pio\build\doitESP32\lib34e\WiFi\WiFiMulti.cpp.o
Compiling .pio\build\doitESP32\lib34e\WiFi\WiFiSTA.cpp.o
Compiling .pio\build\doitESP32\lib34e\WiFi\WiFiScan.cpp.o
Compiling .pio\build\doitESP32\lib34e\WiFi\WiFiServer.cpp.o
Compiling .pio\build\doitESP32\lib34e\WiFi\WiFiUdp.cpp.o
Assembler messages:
Fatal error: can't create .pio\build\doitESP32\src\RFLink.ino.cpp.o: No such file or directory
*** [.pio\build\doitESP32\src\RFLink.ino.cpp.o] Error 1
Assembler messages:
Fatal error: can't create .pio\build\doitESP32\libecd\PubSubClient\PubSubClient.cpp.o: No such file or directory
*** [.pio\build\doitESP32\libecd\PubSubClient\PubSubClient.cpp.o] Error 1
```

if i uncomment WiFiLib i've got

```
Linking .pio\build\doitESP32\firmware.elf
.pio\build\doitESP32\src\6_WiFi_MQTT.cpp.o:(.literal._ZN6RFLink4Mqtt9reconnectEib+0x30): undefined reference to `RFLink::Wifi::stop_WIFI()'
.pio\build\doitESP32\src\6_WiFi_MQTT.cpp.o:(.literal._ZN6RFLink4Mqtt9reconnectEib+0x34): undefined reference to `RFLink::Wifi::start_WIFI()'
.pio\build\doitESP32\src\6_WiFi_MQTT.cpp.o: In function `RFLink::Mqtt::reconnect(int, bool)':
D:\portapps\vs-projects\RFLink\RFLink-Interrupt/RFLink/6_WiFi_MQTT.cpp:105: undefined reference to `RFLink::Wifi::stop_WIFI()'
D:\portapps\vs-projects\RFLink\RFLink-Interrupt/RFLink/6_WiFi_MQTT.cpp:105: undefined reference to `RFLink::Wifi::start_WIFI()'
```

cpainchaud, 2021-02-14:
Hi,

 Reclone my repo and retry, i am sorry for the mess it is right now!

cpainchaud, 2021-02-14:
it does compile on a blan machine:
https://github.com/cpainchaud/RFLink/runs/1898095168

Dattel, 2021-02-14:
out of the box for me NO SUCCESS:

Compiling as it is (except changing the boardtype and serial connection speed)
```
.pio\build\doitESP32\src\6_WiFi_MQTT.cpp.o:(.literal._ZN6RFLink4Mqtt9reconnectEib+0x30): undefined reference to `RFLink::Wifi::stop_WIFI()'
.pio\build\doitESP32\src\6_WiFi_MQTT.cpp.o:(.literal._ZN6RFLink4Mqtt9reconnectEib+0x34): undefined reference to `RFLink::Wifi::start_WIFI()'
.pio\build\doitESP32\src\6_WiFi_MQTT.cpp.o: In function `RFLink::Mqtt::reconnect(int, bool)':
D:\portapps\vs-projects\RFLink\RFLink-Interrupt/RFLink/6_WiFi_MQTT.cpp:105: undefined reference to `RFLink::Wifi::stop_WIFI()'
D:\portapps\vs-projects\RFLink\RFLink-Interrupt/RFLink/6_WiFi_MQTT.cpp:105: undefined reference to `RFLink::Wifi::start_WIFI()'
```

Compiling with RFLINK_WIFIMANAGER_ENABLED or RFLINK_WIFI_ENABLED  :
```
.pio\build\doitESP32\src\6_WiFi_MQTT.cpp.o:(.bss.WIFI_SSID+0x0): multiple definition of `WIFI_SSID'
.pio\build\doitESP32\src\10_Wifi.cpp.o:(.bss.WIFI_SSID+0x0): first defined here
.pio\build\doitESP32\src\6_WiFi_MQTT.cpp.o:(.bss.WIFI_PSWD+0x0): multiple definition of `WIFI_PSWD'
.pio\build\doitESP32\src\10_Wifi.cpp.o:(.bss.WIFI_PSWD+0x0): first defined here
```

cpainchaud, 2021-02-14:
Yes I found out that MQTT was enabled by default and issued a patch a few
minutes ago. Use my latest latest master release and make sure you use the
build flags so you dont get annoyed:

;   -D RFLINK_WIFIMANAGER_ENABLED
;   -D RFLINK_WIFI_ENABLED
;   -D RFLINK_SHOW_CONFIG_PORTAL_PIN_BUTTON=22
;   -D RFLINK_OTA_ENABLED
;   -D RFLINK_OTA_PASSWORD='"'${sysenv.OTA_SEC}'"'
    -D RFLINK_ASYNC_RECEIVER_ENABLED
;   -D MQTT_ENABLED
;   -D RFLINK_SERIAL2NET_ENABLED
;   -D RFLINK_SERIAL2NET_DEBUG

Dattel, 2021-02-14:
👍 compiles and uploads successfully... didn't test on any sensory, yet....

cpainchaud, 2021-03-18:
Hi, did you have any luck? On my side We have done a lot of work on the project so it's more usable and now firmware can be downloaded as binaries and is configurable via a WEB UI.

Let me know if you have time to work on this

Dattel, 2021-03-21:
not really..... 
ESP laying around in my cupboard :-D

I'm using with SignalESP and FHEM for a long time to decode sensors - that work's pretty well and all of my sensors are recognized.
Since the backpart of SignalESP i very strong linked with the FHEM structure i was looking for another ESP-FW, which does the decoding by itself instead of pushing the raw-values to another backend. 

My intention was to be a bit more free to (maybe) test Homeassistant or other systems with an independend ESP-Device. Sadly, there is no system, which is able to decode my sensors out of the box - so my project is currently on ice..