couin3/RFLink #44: Add support for Sonoff T2EU3C-RF

Dacesilian, 2021-01-04:
Hello, thanks for your work on this library!

We have 3-button Sonoff T2EU3C-RF and it's working only partly:
- left button not detected
- center and right button detected as:
20;CE;Eurodomest;ID=0f1bff;SWITCH=05;CMD=ALLON;
20;CF;Eurodomest;ID=0f1bff;SWITCH=06;CMD=ALLON;

Is it possible to add support for this device? https://www.itead.cc/sonoff-t2eu-rf-86-type-433-wall-sticky-rf-remote.html
Thank you.

cpainchaud, 2021-01-29:
Hi ! 

Can you please run this debug command in RFLINK:      10;rfdebug=on; 
Then click your button to capture it on RFlink  and copy/paste the output here.

Thank you !

Dacesilian, 2021-02-17:
@cpainchaud Hello, there are left-center-right button-outputs.

Can you please tell me in short, how you would implement new device based on this output? I think you will "simply" find a repeated pattern in it? How long does it need to be, what part to select? I would like to implement more devices, if I know how to. Thank you.

```
20;XX;DEBUG;Pulses=50;Pulses(uSec)=320,544,288,544,288,544,288,544,672,128,672,160,672,160,256,512,320,544,672,160,256,544,256,544,256,544,256,544,256,544,256,512,320,544,256,544,256,544,256,544,672,160,256,576,256,544,256,608,256,448;
RF;67;esp_rflink;DEBUG;
20;XX;DEBUG;Pulses=42;Pulses(uSec)=640,160,640,160,640,160,256,512,288,576,640,160,256,576,256,576,256,576,256,576,256,576,256,512,288,576,256,576,256,576,256,576,640,160,256,576,256,576,256,608,256,160;
RF;68;esp_rflink;DEBUG;
20;XX;DEBUG;Pulses=44;Pulses(uSec)=256,576,640,192,640,160,640,160,256,512,288,576,640,192,256,576,256,576,256,576,256,576,256,576,256,544,288,576,256,576,256,576,224,576,640,192,224,576,224,576,224,640,224,576;
RF;69;esp_rflink;DEBUG;
20;XX;DEBUG;Pulses=44;Pulses(uSec)=224,576,640,192,640,192,640,192,224,544,288,576,640,192,224,576,224,576,224,576,224,576,224,576,224,544,288,576,224,576,224,576,224,576,640,192,224,576,224,576,224,640,224,576;
RF;6A;esp_rflink;DEBUG;
20;XX;DEBUG;Pulses=44;Pulses(uSec)=224,576,640,192,640,192,640,192,224,544,288,576,640,192,224,576,224,576,224,576,224,576,224,576,224,544,288,576,224,576,224,576,224,576,640,192,224,576,224,576,224,640,224,576;
RF;6B;esp_rflink;DEBUG;
20;XX;DEBUG;Pulses=44;Pulses(uSec)=224,576,640,192,640,192,640,192,224,544,288,576,640,192,224,576,224,576,224,576,224,576,224,576,224,544,288,576,224,576,224,576,224,576,640,192,224,576,224,576,224,640,224,576;
RF;6C;esp_rflink;DEBUG;
20;XX;DEBUG;Pulses=44;Pulses(uSec)=224,576,640,192,640,192,640,192,224,544,288,576,640,192,224,576,224,576,224,576,224,576,224,576,224,544,288,576,224,576,224,576,224,576,640,192,224,576,224,576,224,640,224,576;
RF;6D;esp_rflink;DEBUG;
20;XX;DEBUG;Pulses=44;Pulses(uSec)=224,576,640,192,640,192,640,192,224,544,288,576,640,192,224,576,224,576,224,576,224,576,224,576,224,544,288,576,224,576,224,576,224,576,640,192,224,576,224,576,224,640,224,576;
RF;6E;esp_rflink;DEBUG;
20;XX;DEBUG;Pulses=42;Pulses(uSec)=224,576,640,192,640,192,640,192,224,544,288,576,640,192,224,576,224,576,224,576,224,576,224,576,224,544,288,576,224,576,224,576,224,576,640,192,224,576,224,576,224,640;
RF;6F;esp_rflink;DEBUG;

20;XX;DEBUG;Pulses=50;Pulses(uSec)=320,544,256,544,288,544,288,544,672,160,672,160,672,160,256,512,320,544,672,128,256,544,288,544,256,544,256,544,256,544,256,512,288,576,256,544,256,576,256,544,256,544,640,160,256,576,256,608,256,1952;
RF;A2;esp_rflink;DEBUG;
20;XX;DEBUG;Pulses=42;Pulses(uSec)=640,160,672,160,640,160,256,512,288,576,640,160,256,576,256,576,256,576,256,576,256,576,256,544,288,576,256,576,256,576,256,576,256,576,640,160,256,576,256,640,256,544;
RF;A3;esp_rflink;DEBUG;
20;XX;DEBUG;Pulses=44;Pulses(uSec)=256,576,640,160,640,160,640,160,256,512,288,576,640,160,256,576,256,576,224,576,224,576,224,576,256,544,288,576,224,576,224,576,224,576,224,576,640,192,224,576,224,640,224,160;
RF;A4;esp_rflink;DEBUG;
20;XX;DEBUG;Pulses=44;Pulses(uSec)=224,576,640,192,640,192,640,192,224,544,288,576,640,192,224,576,224,576,224,576,224,576,224,576,224,544,288,576,224,576,224,576,224,576,224,576,640,192,224,576,224,640,224,160;
RF;A5;esp_rflink;DEBUG;
20;XX;DEBUG;Pulses=44;Pulses(uSec)=224,576,640,192,640,192,640,192,224,544,288,576,640,192,224,576,224,576,224,576,224,576,224,576,224,544,288,576,224,576,224,576,224,576,224,576,640,192,224,576,224,640,224,160;
RF;A6;esp_rflink;DEBUG;
20;XX;DEBUG;Pulses=44;Pulses(uSec)=224,576,640,192,640,192,640,192,224,544,288,576,640,192,224,576,224,576,224,576,224,576,224,576,224,544,288,576,224,576,224,576,224,576,224,576,640,192,224,576,224,640,224,160;
RF;A7;esp_rflink;DEBUG;
20;XX;DEBUG;Pulses=44;Pulses(uSec)=224,576,640,192,640,192,640,192,224,544,288,576,640,192,224,576,224,576,224,576,224,576,224,576,224,544,288,576,224,576,224,576,224,576,224,576,640,192,224,576,224,640,224,160;
RF;A8;esp_rflink;DEBUG;
20;XX;DEBUG;Pulses=46;Pulses(uSec)=224,576,224,576,640,192,640,192,640,192,224,544,288,576,640,192,224,576,224,576,224,576,224,576,224,576,224,544,288,576,224,576,224,576,224,576,224,576,640,192,224,576,224,640,224,576;
RF;A9;esp_rflink;DEBUG;
20;XX;DEBUG;Pulses=44;Pulses(uSec)=224,576,224,576,640,192,640,192,640,192,224,544,288,576,640,192,224,576,224,576,224,576,224,576,224,576,224,544,288,576,224,576,224,576,224,576,224,576,640,192,224,576,224,640;
RF;AA;esp_rflink;DEBUG;

20;XX;DEBUG;Pulses=50;Pulses(uSec)=320,544,288,544,288,544,288,544,672,128,672,160,672,160,288,480,320,544,672,160,288,544,288,544,256,576,256,544,256,576,256,512,320,544,256,544,256,544,256,576,256,576,256,544,672,160,256,608,256,1952;
RF;D5;esp_rflink;DEBUG;
20;XX;DEBUG;Pulses=42;Pulses(uSec)=640,160,640,160,672,160,256,512,288,576,640,160,256,576,256,576,256,576,256,576,256,576,256,512,288,576,256,576,256,576,256,576,256,576,256,576,640,192,256,608,256,576;
RF;D6;esp_rflink;DEBUG;
20;XX;DEBUG;Pulses=44;Pulses(uSec)=256,576,640,160,640,160,640,160,256,512,288,576,640,160,256,576,224,576,224,576,224,576,224,576,256,544,288,576,256,576,224,576,224,576,224,576,224,576,640,192,256,640,256,544;
RF;D7;esp_rflink;DEBUG;
[... cut ...] 224,544;
RF;DB;esp_rflink;DEBUG;
20;XX;DEBUG;Pulses=44;Pulses(uSec)=224,576,640,192,608,192,608,192,224,544,256,608,640,192,224,608,224,608,224,608,224,608,224,608,224,544,256,608,224,608,224,608,224,608,224,608,224,608,608,192,224,640,224,544;
RF;DC;esp_rflink;DEBUG;
20;XX;DEBUG;Pulses=42;Pulses(uSec)=224,608,640,192,640,192,640,192,224,544,256,608,608,192,224,608,224,608,224,608,224,608,224,608,224,544,256,608,224,608,224,608,224,608,224,608,224,608,608,192,224,640;
RF;DD;esp_rflink;DEBUG;
```

cpainchaud, 2021-02-18:
Hi!

We need to know what action you did before you capture. Example:

- Press button #1
20;XX;DEBUG;Pulses=50;Pulses(uSec)=352,512,288,544,288,544,288,544,672,160,672,128,672,160,256,512,320,544,672,160,256,544,256,544,256,576,256,544,256,544,256,512,320,544,256,544,256,544,288,544,256,544,256,544,640,160,256,608,256,1952;
RF;58;esp_rflink_vendy_1;DEBUG;
20;XX;DEBUG;Pulses=42;Pulses(uSec)=640,160,640,160,640,160,256,512,288,576,640,160,256,576,256,576,256,576,256,576,256,576,256,512,288,576,256,576,256,576,224,576,256,576,256,576,640,192,256,640,224,544;
RF;59;esp_rflink_vendy_1;DEBUG;

- Press button 2
20;XX;DEBUG;Pulses=44;Pulses(uSec)=224,608,640,192,640,192,608,192,224,544,256,608,608,192,224,608,224,608,224,608,224,608,224,576,224,544,256,608,224,608,224,608,224,608,224,576,224,608,608,192,224,640,224,544;
RF;5F;esp_rflink_vendy_1;DEBUG;
20;XX;DEBUG;Pulses=42;Pulses(uSec)=224,576,640,192,640,192,640,192,224,544,256,608,608,192,224,608,224,608,224,608,224,608,224,608,224,544,256,608,224,608,224,608,224,608,224,608,224,608,608,192,224,640;
RF;60;esp_rflink_vendy_1;DEBUG;

- Press button 3
20;XX;DEBUG;Pulses=44;Pulses(uSec)=224,608,640,192,640,192,608,192,224,544,256,608,608,192,224,608,224,608,224,608,224,608,224,576,224,544,256,608,224,608,224,608,224,608,224,576,224,608,608,192,224,640,224,544;
RF;5F;esp_rflink_vendy_1;DEBUG;
20;XX;DEBUG;Pulses=42;Pulses(uSec)=224,576,640,192,640,192,640,192,224,544,256,608,608,192,224,608,224,608,224,608,224,608,224,608,224,544,256,608,224,608,224,608,224,608,224,608,224,608,608,192,224,640;
RF;60;esp_rflink_vendy_1;DEBUG;

Dacesilian, 2021-02-18:
@cpainchaud Of course - it's divided for left, center, right button press.

cpainchaud, 2021-02-18:
ok, never mind! so it is sending several packets each time? or is it the response of other devices that we see after that ?

Dacesilian, 2021-02-18:
@cpainchaud I'm quite sure that it's only from this devices, not any others (I've done it multiple times). So probably yes, multiple packets each time.

cpainchaud, 2021-02-18:
it is fairly unusual, usually devices sent 1 message (and replay it 2 or 3 times) , here we see many different messages. What are the other devices you are trying to interract with? bulbs?

cpainchaud, 2021-02-18:
may be I found something, do you have a second device or only 1 ? So i can capture the differences ...

Dacesilian, 2021-02-18:
Sadly, only one of this device.

cpainchaud, 2021-02-18:
ok, so you want to detect in RFlink, right? you don't need to emulate it,
do you (transmit RF)?

Dacesilian, 2021-02-18:
Not emulate, only detect, yes. Emulating will be maybe second step for some other devices, not this. Thanks.

cpainchaud, 2021-02-18:
ok, because i can potentially create a plugin but many pieces cannot be
confirmed until we get people to send us more codes to confirm what it is
or not  so it may change in the future