couin3/RFLink #61: No data decoded (even in debug mode) on Mega 2560

progman32, 2021-02-20:
Hey, I'm looking to add MQTT-over-Ethernet support for those of us on the AVR platform, as well as adding support for some door and window sensors. However, I've run into a roadblock. Is this fork known to work on the Mega 2560? I can't get my (known good) board to receive anything, even under `RFDEBUG=ON`. Here is a trace of me opening a window on the board running the proprietary upstream R48:
```
20;00;Nodo RadioFrequencyLink - RFLink Gateway V1.1 - R48;
20;01;setGPIO=ON;
20;02;RFDEBUG=ON;
20;03;DEBUG;Pulses=34;Pulses(uSec)=420,930,270,510,270,540,300,510,300,510,270,1050,300,990,270,30,270,1530,270,30,240,60,240,60,270,540,240,1050,270,30,300,510,240,6990;
```

However if I try it on `master` in this fork, I get nothing:
```
20;00;RFLink_ESP;VER=5.1;
Message arrived [Serial] 10;rfdebug=on;
20;01;RFDEBUG=ON;
```

My only changes are `BAUD 57600` and the `sprintf_P` `%s` => `%S` substitution as instructed by the preprocessor error.

Confirmed the receiver board is getting power.

cpainchaud, 2021-02-25:
Hi,

  To be honest, many of us are using ESP8266 et ESP32 which offer natively and a ton of CPU for 2$-5$. Yet we have some ATmega2560 and it works correctly for them. May be the pins are not the same ? did you check Radio.h file for the pinout reference ? Also did you try both Arduino & Master branches?

progman32, 2021-02-26:
Thanks for the info. My primary interest is avoiding Wi-Fi for reliability reasons (though to be fair, there's no reason ethernet support can't be added to ESP32 via an external PHY). I don't see an arduino branch in this repo, is it somewhere else? I did double check the pin assignments (and tried a couple alternate pins just to see).

chm26, 2021-02-26:
Just for info, I am using my ESP with RFLink without Wifi but with USB connection. It is possible.