couin3/RFLink #66: Support for Silvercrest (LIDL) 284705 remote controlled sockets

obones, 2021-03-02:
Hello,

I have a set of 4 remote controlled sockets from Silvercrest, which IAN is 284705. The remote has 4 pairs for each socket plus one "master" pair that controls them all.

Using the debug mode, I was able to get pulses when each button is pressed. Basically, there is one long frame followed by four shorter frames. Maybe those four shorter frames are a reply by each socket. In any case, here is the dump:

```
A->On
20;XX;DEBUG;Pulses=200;Pulses(uSec)=1248,288,416,1088,1152,352,1152,352,1152,384,1120,384,352,1120,1120,384,1120,384,352,1152,352,1152,1120,416,1120,384,1120,416,320,1184,352,1152,352,1152,1120,416,320,1184,1088,384,1120,416,1088,416,320,1184,320,1184,352,2304,1088,416,320,1152,1088,416,1088,416,1088,416,1088,416,320,1184,1088,416,1088,416,320,1184,320,1184,1088,416,1088,416,1088,416,320,1184,320,1184,320,1184,1088,416,320,1152,1088,416,1088,416,1088,416,320,1184,320,1184,320,2336,1088,416,320,1184,1088,416,1088,448,1056,416,1088,416,320,1184,1088,448,1088,416,320,1184,320,1184,1088,416,1088,448,1088,448,288,1184,320,1216,288,1184,1088,416,320,1184,1088,416,1088,448,1056,448,320,1216,288,1216,320,2336,1088,448,288,1184,1088,448,1056,448,1056,448,1056,448,288,1216,1056,448,1056,448,288,1216,288,1216,1056,448,1056,448,1056,448,288,1216,288,1216,288,1216,1056,448,288,1216,1056,448,1056,448,1056,448,288,1216,288,1216,2976,480;
20;XX;DEBUG;Pulses=36;Pulses(uSec)=928,576,928,576,416,1088,416,1088,928,576,928,608,928,608,416,1088,416,1120,384,1088,928,608,384,1120,896,576,928,576,928,608,416,1088,416,1088,2976,416;
20;XX;DEBUG;Pulses=48;Pulses(uSec)=416,1088,928,576,928,576,928,576,928,576,416,1088,928,576,928,576,416,1088,416,1088,928,576,928,608,896,576,416,1088,416,1120,384,1120,896,672,352,1152,896,640,896,608,896,640,384,1120,384,1152,2976,1184;
20;XX;DEBUG;Pulses=46;Pulses(uSec)=928,608,896,608,896,608,928,608,384,1120,928,608,896,608,384,1120,416,1120,928,608,928,608,896,608,384,1120,384,1120,384,1120,928,608,384,1120,928,608,928,608,896,608,384,1120,384,1120,2976,1152;
20;XX;DEBUG;Pulses=44;Pulses(uSec)=896,608,928,608,928,608,928,608,384,1120,928,608,928,608,384,1120,384,1120,928,608,896,608,896,608,384,1120,384,1120,384,1120,896,608,384,1120,896,608,896,608,928,608,384,1120,416,1120;

A->Off
20;XX;DEBUG;Pulses=202;Pulses(uSec)=128,608,1184,320,384,1120,1184,352,1152,352,1120,384,352,1152,352,1152,1120,384,1120,416,1120,416,320,1184,1120,416,1088,416,320,1184,320,1184,352,1152,1120,416,320,1184,320,1184,320,1184,1088,416,1088,448,288,1184,320,1184,320,2368,1088,448,288,1216,1088,448,1088,448,1088,448,288,1216,320,1184,1088,448,1088,448,1088,416,320,1184,1088,448,1056,448,288,1184,320,1216,320,1216,1056,448,288,1216,320,1216,288,1184,1088,416,1088,448,320,1184,320,1184,320,2336,1088,448,320,1184,1088,448,1056,416,1088,448,288,1184,320,1184,1088,448,1088,448,1056,448,320,1184,1088,416,1088,448,320,1184,320,1184,320,1184,1088,448,320,1184,320,1184,320,1184,1088,416,1088,448,288,1216,320,1216,320,2336,1088,448,288,1216,1088,448,1088,448,1056,448,320,1216,320,1216,1088,448,1056,448,1056,448,288,1216,1056,448,1056,448,288,1216,288,1216,288,1216,1088,448,288,1216,288,1216,288,1216,1056,448,1056,448,288,1216,288,1248,2976,96;
20;XX;DEBUG;Pulses=36;Pulses(uSec)=928,576,928,576,928,576,416,1120,896,608,928,576,416,1088,416,1088,416,1120,928,608,416,1120,416,1120,384,1120,928,576,928,608,384,1120,384,1120,2976,416;
20;XX;DEBUG;Pulses=48;Pulses(uSec)=416,1088,928,608,896,576,928,608,384,1120,416,1120,928,608,896,608,928,608,384,1120,928,608,896,608,416,1120,416,1120,416,1120,928,608,384,1120,416,1120,416,1120,928,608,896,608,384,1120,384,1120,2976,1184;
20;XX;DEBUG;Pulses=46;Pulses(uSec)=928,576,928,576,928,608,416,1120,416,1120,928,608,928,608,928,608,384,1120,928,608,896,608,384,1120,416,1120,384,1120,928,608,384,1120,384,1120,384,1120,928,640,896,608,384,1120,384,1152,2976,1120;
20;XX;DEBUG;Pulses=44;Pulses(uSec)=896,608,928,608,928,608,384,1120,384,1120,928,608,896,608,896,608,384,1120,928,608,896,608,384,1120,384,1120,416,1120,896,608,384,1120,384,1120,384,1120,896,608,896,640,384,1120,384,1152;

B->On
20;XX;DEBUG;Pulses=196;Pulses(uSec)=1184,320,1152,352,384,1152,352,1120,384,1120,384,1152,1120,448,1088,416,1120,384,1120,384,352,1152,1120,416,1088,416,320,1216,1088,448,1056,416,1088,448,320,1184,320,1184,1088,416,320,1216,1088,448,320,2336,1088,416,320,1184,1088,416,1088,448,320,1184,320,1184,320,1184,320,1184,1088,448,1088,448,1088,448,1056,416,320,1216,1088,448,1088,448,288,1184,1088,448,1056,448,1088,448,288,1216,320,1216,1088,448,288,1216,1088,448,288,2336,1088,448,320,1216,1088,448,1056,448,288,1216,320,1216,288,1216,320,1216,1088,448,1056,480,1024,448,1056,448,288,1216,1056,480,1056,448,288,1216,1056,448,1056,480,1056,480,288,1216,288,1216,1088,448,288,1216,1088,480,288,2368,1088,448,288,1216,1056,448,1056,448,288,1216,288,1216,320,1248,288,1216,1088,448,1056,448,1056,480,1056,480,256,1248,1056,480,1056,448,288,1248,1056,480,1056,480,1056,480,256,1248,288,1216,1056,480,288,1248,1056,480,2976,448;
20;XX;DEBUG;Pulses=36;Pulses(uSec)=416,1120,928,608,928,608,896,608,928,608,416,1120,928,608,896,672,352,1120,928,608,896,608,896,608,416,1120,416,1120,928,576,416,1088,928,608,2976,1184;
20;XX;DEBUG;Pulses=48;Pulses(uSec)=416,1056,960,576,928,576,416,1088,416,1088,416,1088,416,1088,928,576,928,576,928,576,928,608,384,1088,928,576,928,576,416,1088,928,576,928,608,896,576,416,1088,416,1088,928,576,416,1120,896,576,2976,416;
20;XX;DEBUG;Pulses=46;Pulses(uSec)=928,576,928,576,416,1088,416,1088,416,1088,416,1088,928,576,928,576,928,608,896,608,416,1088,928,576,928,576,416,1088,928,576,928,576,928,576,416,1120,384,1088,928,608,416,1088,928,608,2976,576;
20;XX;DEBUG;Pulses=44;Pulses(uSec)=928,576,928,576,416,1088,416,1088,416,1088,416,1088,928,576,928,608,928,608,928,576,416,1088,928,608,896,608,416,1088,928,608,896,608,928,608,384,1120,384,1088,928,608,416,1088,928,608;

B->Off
20;XX;DEBUG;Pulses=196;Pulses(uSec)=1152,352,1152,384,352,1120,352,1152,1152,352,1120,384,1120,384,352,1152,1120,384,1120,384,1120,384,352,1184,320,1152,1120,384,1120,416,320,1152,320,1184,1088,416,320,1184,1120,416,320,1184,1088,416,320,2336,1088,416,320,1184,1088,416,1088,416,320,1184,320,1184,1088,416,1088,416,1088,416,320,1184,1088,416,1088,416,1088,416,320,1184,320,1184,1088,416,1088,416,320,1184,320,1184,1088,416,320,1184,1088,416,320,1184,1088,416,320,2336,1088,416,320,1184,1088,416,1088,448,288,1184,320,1184,1088,416,1088,416,1088,416,320,1184,1088,416,1088,416,1088,448,288,1216,288,1184,1088,448,1088,448,288,1216,288,1184,1088,448,288,1184,1088,448,288,1184,1088,448,288,2336,1056,448,288,1184,1088,416,1088,448,288,1216,288,1216,1056,448,1056,448,1056,448,288,1184,1088,448,1056,448,1056,448,288,1216,288,1216,1056,448,1056,448,288,1216,288,1216,1056,448,288,1216,1056,448,288,1216,1056,448,2976,448;
20;XX;DEBUG;Pulses=36;Pulses(uSec)=928,576,928,576,416,1088,928,576,928,576,928,576,416,1088,416,1088,928,576,928,608,416,1120,416,1120,896,640,384,1120,928,576,416,1088,928,608,2976,416;
20;XX;DEBUG;Pulses=48;Pulses(uSec)=416,1088,928,576,928,576,416,1088,416,1088,928,576,928,576,928,608,416,1088,928,576,928,608,928,608,384,1120,416,1120,928,608,928,608,384,1120,416,1120,928,608,416,1120,928,608,384,1120,928,608,2976,416;
20;XX;DEBUG;Pulses=46;Pulses(uSec)=928,608,928,576,416,1088,416,1088,928,608,928,576,928,576,416,1120,928,576,928,608,896,608,416,1088,416,1088,928,608,928,608,416,1120,384,1120,928,608,416,1088,928,608,384,1120,896,608,2976,608;
20;XX;DEBUG;Pulses=44;Pulses(uSec)=928,576,928,576,416,1088,416,1088,928,576,928,576,928,608,416,1120,928,608,896,608,896,608,384,1120,416,1088,928,608,928,608,384,1120,416,1120,928,608,384,1120,928,608,384,1120,928,608;

C->On
20;XX;DEBUG;Pulses=200;Pulses(uSec)=1152,352,384,1120,1152,384,1120,384,384,1120,1152,384,352,1120,1120,384,384,1120,352,1152,1120,384,352,1152,352,1152,352,1152,352,1152,1120,384,352,1152,352,1184,1120,416,352,1152,1120,384,1120,384,1120,416,320,1152,352,2336,1088,416,352,1184,1088,416,1088,416,320,1184,1088,416,320,1184,1088,416,320,1184,320,1184,1088,416,320,1184,320,1184,320,1184,320,1184,1088,416,320,1184,320,1184,1088,416,320,1184,1088,416,1088,416,1088,416,320,1184,320,2336,1120,416,320,1184,1088,416,1088,448,320,1184,1088,448,288,1184,1088,416,320,1184,320,1184,1088,448,320,1184,320,1184,320,1184,320,1216,1088,448,288,1184,320,1216,1088,416,320,1184,1088,416,1088,448,1056,416,320,1216,320,2336,1088,448,288,1184,1088,448,1088,448,288,1184,1088,448,288,1216,1056,448,288,1216,288,1184,1 [... cut ...]
20;XX;DEBUG;Pulses=46;Pulses(uSec)=928,576,928,576,928,576,928,576,928,576,928,576,416,1088,928,576,928,576,928,576,416,1088,928,576,416,1088,928,576,416,1088,928,576,928,576,416,1120,416,1088,416,1088,928,608,384,1120,2976,1120;
20;XX;DEBUG;Pulses=42;Pulses(uSec)=928,576,928,576,928,576,928,576,928,576,416,1088,928,608,928,576,928,576,416,1088,928,576,416,1088,928,576,416,1088,928,576,928,576,416,1120,416,1088,416,1088,928,608,416,608;
```

The original RFLink claims to support them, but it does not appear in this fork.
How can I help to analyze these messages?

Just in case, here are photos of the remote:
[image]
[image]

obones, 2021-03-03:
Ok, unpluggin one socket does not change the number of debug packets seen by the RFLink, so I believe they are part of the message sent by the remote

obones, 2021-03-05:
This repository is sending messages to similar remotes: https://github.com/Mr-Codesaur/SilverCrest_433Mhz_Sockets-Arduino/blob/master/SilverCrest_Sockets.h

It shows what I have also found by myself, the message is 24 bits long, and the above trace shows it 8 times. The first 4 are with a short time base, while the 4 last are with a longer time base. But basically, the way it works is a preamble (Short high, very long low) followed by 24 combinations of either Long High / Short Low  or Short High / Long Low giving 1 or 0 as the bit value.

Pressing the same remote button however gives 4 different possible values. For instance, for the On button on row A, we get this:

value | byte 2 | byte 1 | byte 0
-- | -- | -- | --
bf756c | 10111111 | 01110101 | 01101100
b7441c | 10110111 | 01000100 | 00011100
bd9c5c | 10111101 | 10011100 | 01011100
be3aac | 10111110 | 00111010 | 10101100

And comparing with other buttons, we get to understand that the first 4 bits are always the same, the last 4 bits define which button is pressed and never change among the 4 possible values.
But the remaining 16 bits are changing and I cannot yet spot the place where it says On or Off 

For reference, here are the values:

A->On | A->Off | B->On | B->Off | C->On | C->Off | D->On | D->Off | Master->On | Master->Off
-- | -- | -- | -- | -- | -- | -- | -- | -- | --
bf756c | b20e7c | b6af35 | b16bc5 | b5212e | be3aae | bae247 | bc1707 | b9d882 | bd9c52
b7441c | b5212c | b453f5 | b3b995 | b9d88e | bf756e | b16bc7 | b6af37 | b52122 | b74412
bd9c5c | b9d88c | b0f6e5 | bb8dd5 | b8c0be | b7441e | b3b997 | b453f7 | b20e72 | bf7562
be3aac | b8c0bc | bc1705 | bae245 | b20e7e | bd9c5e | bb8dd7 | b0f6e7 | b8c0b2 | be3aa2

So, `b` is the constant part at the beginning, most likely a remote id
And the buttons have the following code in the last 4 bits:

A|B|C|D|Master
--|-- |-- |-- |-- 
c|5|e|7|2

obones, 2021-03-06:
When we look at the middle 16 bits, we see that the patterns are used either for an On or an Off command, depending on the button. 
The truth table is like this:

  | On | Off
-- | -- | --
f756 | A | C  <br/>  Master
7441
d9c5
e3aa
20e7 | C <br/> Master | A
5212
9d88
8c0b
6af3 | B | D
453f
0f6e
c170
16bc | D | B
3b99
b8dd
ae24

So, the same `b8dd` code is either an On or a Off depending on the button being pressed. Interestingly, this inversion of meaning is given by the value of bit 1 inside the button Id.
If we split the command values in two arrays of 8 values containing the On or Off values, bit 0 inside the button Id tells us in which part of the truth table we must be looking for.

After experimenting, the sockets react to any of the four possible values, it does not have to be the next one in the series everytime, so clearly no rolling code here.

This means it is possible to create a very simple plugin for this.

simonhyde, 2022-09-16:
I've been looking at why RFLink32 on a Sonoff RF Bridge doesn't seem to work with my Silvercrest 284705 remotes/sockets, and it looks like the middle bit patterns and button IDs change with different remoteIds. So there may be 16 different sets of bit patterns:

So, for remoteId 0xD:

| A | B | C | D | Master |
| - | - | - | - | -------- |
| 0 | 4 | C | 2 | A |

And the middle bit patterns:

|     | On | Off |
| --- | --- | --- |
| cd63 | A C B | D Master |
| f7b4 | | |
| b4f7 | | |
| 4ceb | | |
| ae81 | D Master | A C B |
| 8159 | | |
| 782a | | |
| 9292 | | |

Note there only seem to be 8 possible codes on this remote, not the 16 you had, which I think makes sense because all the remote buttons are even...

Adding those 8 codes to the relevant On/Off tables makes it possible to receive codes, transmitting is obviously a little more fiddly, without having all 16 possible remotes I'm not clear on how we'd determine what the possible middle bits are...For now I've simply bodged the Tx code to always use codeIndex of 9 if the remoteId is d

It looks like the second bit of buttonId is still determining whether the codes on/off meaning should be inverted.

obones, 2022-09-16:
I actually have bought a second set of remote controlled sockets from LIDL and while they share the same structure, they do not have the same middle two bytes as the one in this PR, nor the same as yours.
So there must definitely be a rule as to how to compute those two bytes, but I'm yet to find out which one.
Considering the size, it could well be a CRC16 from two other bytes : a constant and a counter that goes from 0 to 3 for instance, along maybe with another bit that says "master command".
One way to figure that out would be to test for all possible CRC16 values from all possible values in one or two bytes. 
I have in my never ending to do list an item to use [reveng](https://sourceforge.net/projects/reveng/) to do just this, I'm just lacking the time to do so...

simonhyde, 2023-11-06:
So, for remoteId 0x5:

| A | B | C | D | Master |
| - | - | - | - | -------- |
| 0 | 4 | C | 2 | A |

And the middle bit patterns:

|     | On | Off |
| --- | --- | --- |
| fa1b | A C B | D Master |
| 3da3 | | |
| 0254 | | |
| 8c27 | | |
| 6e41 | D Master | A C B |
| 1bb9 | | |
| a3fa | | |
| c02 | | |

There's an extra 0 in the final code, but I'm not sure exactly where it sits, because there's a bug in the way the debug printing works, it prints a single digit for a byte when given a value < 0x10