couin3/RFLink #74: No output / not working?

mamartins, 2021-04-04:
Hello,

I'm trying to decode my awnings RF codes, I'm using a Wemos D1 and a RXB6 receiver and I have this PINs config:
```
#ifdef ESP8266
// ESP8266 D1 Mini
#define PIN_RF_RX_PMOS_0 NOT_A_PIN // High Side P-MOSFET, active on LOW level
#define PIN_RF_RX_NMOS_0 NOT_A_PIN        // Low Side N-MOSFET, active on HIGH level
#define PIN_RF_RX_VCC_0 NOT_A_PIN  // Power to the receiver on this pin
#define PIN_RF_RX_GND_0 NOT_A_PIN  // Ground to the receiver on this pin
#define PIN_RF_RX_NA_0 NOT_A_PIN   // Alt. RX_DATA. Forced as input
#define PIN_RF_RX_DATA_0 D7        // On this input, the 433Mhz-RF signal is received. LOW when no signal.
#define PIN_RF_TX_PMOS_0 NOT_A_PIN // High Side P-MOSFET, active on LOW level
#define PIN_RF_TX_NMOS_0 NOT_A_PIN        // Low Side N-MOSFET, active on HIGH level
#define PIN_RF_TX_VCC_0 NOT_A_PIN  // +5 volt / Vcc power to the transmitter on this pin
#define PIN_RF_TX_GND_0 NOT_A_PIN  // Ground power to the transmitter on this pin
#define PIN_RF_TX_NA_0 NOT_A_PIN   // Spare RX pin. Forced as input
#define PIN_RF_TX_DATA_0 D4        // Data to the 433Mhz transmitter on this pin
#endif
```

The debug shows that the device can connect to my WIFI and to MQTT server, but no code is ever sent via MQTT or presented in the terminal. I've also tried with D6 as `PIN_RF_RX_DATA_0`. 

I read the voltage of D7 (and D6 when I connected there) and it's always around 1.2v, as soon as I press one key on the awnings RF transmitter the voltage raises to 1.4 and when I release the key it goes to 0 for a couple of seconds and back to 1.2v. What am I missing?

Thank you

cpainchaud, 2021-04-08:
Hi!
 what model/hardware are you trying to decode?

do you get any output at all with command 10;rfdebug=off;  ?

mamartins, 2021-04-14:
Hello,

I couldn't find any info online regarding the chip other than some Chinese supply sites, it looks like it is a BL101F3P6
[image]

With `RFDebug_0 true` on `RFLink.h` I was able to get this output:
```
20;XX;DEBUG;Pulses=132;Pulses(uSec)=480,672,896,256,480,672,864,288,448,672,480,672,448,672,864,288,448,672,448,672,832,288,832,288,832,288,448,704,448,704,832,320,832,288,832,288,832,320,448,704,416,704,448,704,832,320,448,704,832,320,832,320,448,704,448,704,416,704,448,704,416,704,416,736,416,704,832,320,416,704,416,704,832,320,832,320,416,704,416,736,832,320,800,320,416,704,832,320,416,704,800,320,416,704,800,320,800,320,416,736,416,736,800,320,800,320,800,320,800,320,800,352,416,736,800,320,800,320,800,320,800,320,800,320,416,736,800,352,800,320,800,0;
20;01;DEBUG;
20;XX;DEBUG;Pulses=132;Pulses(uSec)=864,288,480,640,512,640,480,672,864,256,480,672,864,288,448,704,448,672,448,672,448,672,448,672,448,672,864,288,864,288,832,320,832,288,448,704,832,320,448,704,832,288,448,704,832,320,448,704,448,704,832,320,448,704,416,704,448,704,832,320,416,704,416,704,416,704,832,320,416,704,416,704,832,320,832,320,416,736,416,736,832,320,832,320,416,704,800,320,416,704,832,320,416,704,800,320,800,320,416,736,416,736,800,320,800,320,800,320,800,320,800,320,416,736,800,320,800,320,800,320,800,320,800,320,416,736,800,352,800,320,800,0;
20;02;DEBUG;
20;XX;DEBUG;Pulses=132;Pulses(uSec)=864,256,480,672,480,672,864,288,448,704,480,672,448,672,480,672,480,672,448,672,448,672,864,288,448,672,448,704,448,704,832,288,832,288,832,288,448,704,448,704,832,288,448,704,832,320,448,704,448,704,448,704,448,704,416,704,416,704,448,704,832,288,832,320,448,704,832,320,416,704,416,704,832,320,832,320,416,704,416,736,832,320,832,320,416,704,832,320,416,704,800,320,416,704,832,320,800,320,416,736,416,736,800,320,800,320,800,320,800,320,800,320,416,736,800,320,800,320,800,320,800,320,800,320,416,736,800,320,800,320,800,0;
20;03;DEBUG;
20;XX;DEBUG;Pulses=132;Pulses(uSec)=480,640,896,256,480,672,864,288,480,672,864,288,480,672,480,672,864,288,480,672,448,672,864,288,832,288,832,288,832,288,448,704,832,288,448,704,448,704,448,704,832,320,832,320,832,320,832,320,832,320,832,320,832,320,832,320,416,704,416,704,832,320,416,736,416,704,832,320,416,704,416,704,832,320,832,320,416,704,416,736,832,320,832,320,416,704,832,320,416,736,800,320,416,736,800,320,800,320,416,736,416,736,800,320,800,320,800,320,800,320,800,320,416,736,800,320,800,320,800,320,800,320,800,320,416,736,800,352,800,320,800,0;
20;04;DEBUG;
20;XX;DEBUG;Pulses=132;Pulses(uSec)=864,288,480,672,864,288,864,256,864,288,864,288,864,288,448,704,448,672,448,704,448,704,448,672,448,704,832,288,832,288,832,288,832,288,832,288,448,704,416,704,832,320,832,320,832,320,448,704,448,704,448,704,448,704,448,704,416,704,416,704,416,704,832,320,416,704,832,320,416,704,416,704,832,320,832,320,416,704,416,704,832,320,832,320,416,736,800,320,416,704,800,320,416,704,800,320,800,320,416,736,416,736,800,320,800,320,800,320,800,320,800,320,416,736,800,320,800,320,800,320,800,320,800,320,416,736,800,320,800,320,800,0;
20;05;DEBUG;
20;XX;DEBUG;Pulses=132;Pulses(uSec)=864,256,864,288,864,256,480,672,864,288,864,288,864,288,864,288,864,288,864,288,448,704,448,672,832,288,832,288,448,704,832,288,448,704,448,704,832,320,832,288,832,320,448,704,832,320,832,320,448,704,416,704,448,704,416,704,832,320,416,704,416,704,416,736,448,704,832,320,416,704,416,704,832,320,800,320,416,704,416,736,832,320,832,320,416,704,800,320,416,704,832,320,416,704,800,320,800,320,416,736,416,736,800,320,800,320,800,320,800,320,800,320,416,736,800,320,832,320,800,320,800,320,800,320,416,736,800,352,800,320,800,0;
20;06;DEBUG;
20;XX;DEBUG;Pulses=156;Pulses(uSec)=384,320,416,320,416,320,416,320,416,352,416,352,416,352,416,320,416,320,416,352,416,320,416,3904,800,320,800,320,800,320,416,736,800,320,800,352,800,320,800,352,800,352,800,352,416,736,416,736,800,352,800,352,416,736,800,352,416,736,416,736,800,352,800,352,800,320,416,736,800,352,800,352,416,736,416,736,416,736,416,736,800,352,416,736,416,736,416,736,416,736,800,352,416,736,416,736,800,352,800,352,416,736,416,736,800,352,800,352,384,736,800,352,416,736,800,352,384,736,800,352,800,352,384,736,384,736,800,352,800,352,800,352,800,352,800,352,384,736,800,352,800,352,800,352,800,352,800,352,384,736,800,352,800,352,384,0;
20;07;DEBUG;
20;XX;DEBUG;Pulses=132;Pulses(uSec)=512,640,896,224,512,640,512,640,896,256,896,256,896,256,896,256,480,672,864,256,480,672,864,256,480,672,480,672,480,672,864,288,480,672,864,288,480,672,864,288,480,672,864,288,448,672,448,704,448,672,448,672,448,672,448,672,864,288,448,704,448,704,448,704,448,704,832,288,448,704,448,704,832,288,832,288,448,704,448,704,832,320,832,320,448,704,832,320,448,704,832,320,448,704,832,320,832,320,448,704,448,704,832,320,832,320,832,320,832,320,832,320,416,704,832,320,832,320,832,320,832,320,832,320,416,736,800,320,800,320,800,352;
20;08;DEBUG;
20;XX;DEBUG;Pulses=156;Pulses(uSec)=384,320,416,320,416,320,416,320,416,320,416,320,416,320,416,320,416,320,416,320,416,320,416,3904,416,736,800,320,416,736,416,736,800,320,800,352,800,352,800,352,416,736,800,352,416,736,800,352,416,736,416,736,416,736,800,352,384,736,800,352,416,736,800,352,384,736,800,352,384,736,384,736,384,768,384,736,384,736,384,736,800,352,384,736,384,736,384,768,384,736,800,352,384,736,384,736,800,352,800,352,384,736,384,768,800,352,800,352,384,736,768,352,384,768,768,352,384,768,768,352,768,352,384,768,384,768,768,352,768,352,768,352,768,352,768,384,384 [... cut ...]
20;XX;DEBUG;Pulses=156;Pulses(uSec)=352,352,384,352,384,384,384,352,384,352,384,352,384,352,384,352,384,352,384,352,384,352,384,3936,384,768,768,352,384,768,384,768,768,352,768,352,768,352,768,384,384,768,768,384,384,768,768,352,384,768,384,768,384,768,768,384,384,768,768,352,384,768,768,352,384,768,768,352,384,768,384,768,384,768,384,768,384,768,384,768,768,352,384,768,384,768,384,768,384,768,768,384,384,768,384,768,768,352,768,352,384,768,384,768,768,384,768,384,384,768,768,384,384,768,768,384,384,768,768,384,768,384,384,768,384,768,768,384,768,384,768,384,768,384,768,384,384,768,768,384,768,384,768,384,768,384,768,384,384,768,768,384,768,384,384,0;
20;19;DEBUG;
```

cpainchaud, 2021-04-15:
Thank you, this is already helpful. Do you have any link to  store where this item is available for purchase. If we have to write a plugin we need to find a brand to link it to.

And and what is your goal here? do you want to be able to command your awning from your computer/automation or something else? 
I suppose your remote has several buttons (at least one to deploy, one to retract)

Can you show the "Pulses" like you did before but tell us which one are showing after each button so we can compare data and point the differences so we can at least tell which Bits/Pulses are the command ones

By the way you don't need recompile software you can activate debug at runtime with command "10;RFDEBUG=on;"

mamartins, 2021-04-18:
You're welcome

The remote came with the awning, I found it online for ser [here](https://motorline.pt/produto/mx95-mx96-mx97/#dados-tecnicos30de-891b34cf-9b52de02-4a0f56f7-a4714281-a730b266-a4bd8c55-1591), but there isn't any helpful information

My goal is to be able to command my awnings from home assistant based on the wind speed I get from my weather station.

The remote has 5 channels and a "all" channel. Channel 1 control one of the awnings, channel 2 other, and channel "all" controls all at the same time.

These are the readings, by clicking on the remote quickly:
```
OPEN channel 1
20;XX;DEBUG;Pulses=132;Pulses(uSec)=480,672,480,672,480,672,864,256,864,288,864,288,448,672,864,288,448,672,864,288,864,288,448,704,448,672,448,704,448,704,832,320,832,288,832,288,448,704,448,704,448,704,832,320,448,704,448,704,448,704,416,704,448,704,416,704,416,704,832,320,832,320,832,320,416,704,832,320,416,704,416,704,832,320,832,320,416,736,416,736,832,320,832,320,416,736,832,320,416,736,832,320,416,736,832,320,832,320,416,736,416,704,800,320,800,320,800,320,832,320,800,320,416,736,800,320,800,320,800,320,800,320,416,736,800,320,800,320,800,320,800,384;
20;1D;DEBUG;

STOP channel 1
20;XX;DEBUG;Pulses=156;Pulses(uSec)=384,320,416,320,416,320,416,320,416,320,416,320,416,320,416,320,416,320,416,320,416,320,416,3904,416,736,416,736,416,736,416,736,416,736,416,736,800,320,416,736,800,352,800,320,800,320,800,352,416,736,800,352,416,736,416,736,800,352,800,352,416,736,416,736,800,352,416,736,416,736,800,352,800,352,800,352,800,352,800,352,416,736,416,736,800,352,416,736,416,736,800,352,416,736,416,736,800,352,800,352,416,736,416,736,800,352,800,352,416,736,800,352,416,736,800,352,416,736,800,352,800,352,416,736,384,736,800,352,800,352,800,352,800,352,800,352,416,736,800,352,800,352,800,352,800,352,800,352,384,736,800,352,800,352,416,0;
20;1F;DEBUG;

CLOSE channel 1
20;XX;DEBUG;Pulses=132;Pulses(uSec)=864,288,864,288,864,288,448,672,480,672,864,256,448,672,448,704,864,288,448,704,448,672,832,288,448,704,864,288,448,704,832,320,832,320,448,704,832,320,832,320,448,704,448,704,448,704,448,704,448,704,448,704,448,704,832,320,832,320,832,320,416,704,416,704,416,704,832,320,416,704,416,704,832,320,832,320,416,704,416,736,832,320,832,320,416,704,832,320,416,736,832,320,416,736,832,320,800,320,416,704,416,736,800,320,800,320,800,320,800,320,832,320,416,736,800,320,832,320,800,320,416,736,800,320,800,320,800,320,800,320,800,352;
20;21;DEBUG;

OPEN channel 2
20;XX;DEBUG;Pulses=132;Pulses(uSec)=896,256,896,256,480,672,480,672,864,288,448,672,864,288,864,288,864,288,448,672,832,288,448,704,448,704,448,672,832,288,832,288,448,704,832,288,448,704,832,320,448,704,448,704,448,704,832,320,448,704,832,320,448,704,832,288,832,320,832,320,832,320,416,704,416,704,832,320,416,704,416,704,832,320,832,320,416,704,416,736,832,320,832,320,416,704,832,320,416,736,832,320,416,736,800,320,832,320,416,704,416,736,800,320,800,320,800,320,800,320,800,320,416,736,800,320,832,320,800,320,800,320,416,736,800,320,800,320,800,320,800,352;
20;22;DEBUG;

STOP channel 2
20;XX;DEBUG;Pulses=132;Pulses(uSec)=864,256,480,672,864,288,864,288,480,672,480,672,864,288,448,704,864,288,448,672,448,704,448,704,448,704,448,704,448,704,448,704,832,320,832,288,448,704,448,704,832,320,448,704,832,320,832,320,832,288,448,704,832,320,832,320,832,320,832,320,832,320,832,320,416,704,832,320,416,704,416,704,832,320,832,320,416,704,416,736,832,320,832,320,416,704,832,320,416,736,800,320,416,704,800,320,800,320,416,736,416,736,832,320,800,320,800,320,800,320,800,320,416,736,800,320,800,320,800,320,800,320,800,320,416,736,800,352,800,320,800,352;
20;23;DEBUG;

CLOSE channel 2
20;XX;DEBUG;Pulses=132;Pulses(uSec)=864,288,864,256,864,288,864,288,864,288,864,288,448,672,864,288,448,704,448,672,864,288,832,288,448,672,832,288,448,704,832,320,832,320,448,704,832,320,448,704,832,320,832,320,416,704,832,320,448,704,832,320,448,704,448,704,416,704,832,320,416,704,416,736,416,704,832,320,416,704,416,704,832,320,832,320,416,704,416,736,832,320,832,320,416,704,800,320,416,704,832,320,416,736,800,320,832,320,416,736,416,736,800,320,800,320,800,320,800,320,800,320,416,736,800,320,800,320,800,320,416,736,800,320,800,320,800,320,800,320,800,352;
20;24;DEBUG;

OPEN channel all
20;XX;DEBUG;Pulses=132;Pulses(uSec)=448,672,480,672,864,288,864,288,480,672,480,672,832,288,448,704,448,704,448,704,448,704,448,704,832,320,832,288,448,704,448,704,832,320,832,320,832,320,416,704,832,320,416,704,448,704,416,736,416,704,448,704,416,704,832,320,416,704,832,320,832,320,416,736,416,704,832,320,416,736,416,704,832,320,800,320,416,736,416,736,832,320,800,320,416,736,800,320,416,736,800,320,416,736,800,352,800,352,416,736,416,736,800,320,800,320,800,320,800,320,800,352,800,320,800,352,800,320,800,352,800,352,416,736,800,352,800,352,800,352,800,352;
20;25;DEBUG;

STOP channel ALL
20;XX;DEBUG;Pulses=132;Pulses(uSec)=448,672,448,672,480,672,448,672,864,256,448,672,480,672,480,672,448,704,864,288,448,672,448,704,448,704,832,288,832,288,832,320,832,320,832,320,832,320,448,704,448,704,448,704,832,320,832,320,832,320,832,320,832,320,832,320,832,320,832,320,832,320,832,320,416,704,832,320,416,736,416,736,800,320,832,320,416,736,416,736,800,320,800,320,416,736,832,320,416,736,800,320,416,736,800,320,800,320,416,736,416,736,800,320,800,320,800,320,800,352,800,352,800,320,800,352,800,320,800,320,800,320,800,352,416,736,800,352,800,320,800,352;
20;26;DEBUG;

CLOSE channel  ALL
[... cut ...] 864,288,448,704,448,704,480,672,864,288,832,320,448,704,832,320,448,704,832,320,416,704,832,320,416,704,832,320,832,320,416,736,832,320,832,320,416,704,832,320,416,704,832,320,832,320,416,704,416,736,800,320,416,704,416,704,800,320,832,320,416,704,416,736,800,320,800,320,416,736,800,320,416,736,800,320,416,736,800,352,800,320,416,736,416,736,800,320,800,320,800,352,800,352,800,352,800,320,800,320,832,320,800,352,416,736,800,320,800,352,800,352,800,352,800,352;
20;28;DEBUG;
```

I hope this help!

obones, 2021-04-28:
The chip is a 8L101F3P6, a generic 8bit microcontroller from STMicro: https://www.avnet.com/shop/emea/products/stmicroelectronics/stm8l101f3p6-3074457345634470749/

But because it is generic, you would have to extract its Flash memory to be able to figure out what it is sending